projectchrono/chrono #35: checking for unary minus on an unsigned int.

jcmadsen, 2015-02-25:
From ChHashFunction.h, line 77:

```
/// Specialized hash function for strings (other version)
///

struct HashFunction_String2
{
	unsigned operator()(const std::string& key) const 
	{ 
		size_t n = key.length();
		const char* d = key.c_str();
		unsigned h = 0; 
	      
		for (size_t i = 0; i < n; ++i, ++d)
			  h = (h << 2) + *d;

		return ((h >= 0) ? (h ) : (-h ));
	}
};
'''

Question: Why when returning h, which is unsigned, do you check to ensure h is positive? This seems unnecessary.

andrewseidl, 2015-02-25:
@tasora : Is `HashFunction_String2` still needed? The only usage of it that I see is in the auto-generated SWIG wrapper.

rserban, 2015-02-25:
This is something I meant to fix before.  As far as I can tell, these are implementations of GCC hash functions (with the final mod operation done outside, in the calling functions).  I think both `HashFunction_String2` and `HashFunction_String3` should just return the unsigned value `h`.

tasora, 2016-10-19:
Custom hash map not needed, since we moved to C++11 std::unordered_map

closed 2016-10-19 by tasora