projectchrono/chrono #50: demo_FEAbasic test_1() doesn't not successfully complete ChSystem.DoStaticLinear()

jcmadsen, 2015-04-29:
All the GetLog() output results at the end of the test return NaN. 

I've traced the problem to the function ChLcpIterativeMINRES::Solve_SupportingStiffness(ChLcpSystemDescriptor& sysd), and without going into too much detail, the problem occurs after the first few iterations in the loop starting on line 457 in ChIterativeMINRES.cpp, either alpha or beta compute a value of zero in the denominator. 

This also occurs for at least one other example, test_2() in the same demo, and maybe more.

lines 463 - 467:
```
  // alpha = (r' * Zr) / ((Zp)'*(MZp));
        double rZr = r.MatrDot(&r, &Zr);      // 1)  z'* Zr
        double ZpMZp = r.MatrDot(&Zp, &MZp);  // 2)  ZpMZp = (Zp)'*(MZp)

        double alpha = rZr / ZpMZp;
```

lines 498 - 502:

```
        // beta = (r' * N*r) / (r_old' * N*r_old);
        double numerator = r.MatrDot(&r, &Zr);            // 1)  r'* Z *r
        double denominator = r.MatrDot(&r_old, &Zr_old);  // 2)  r_old'* Z *r_old

        double beta = numerator / denominator;
```

jcmadsen, 2015-04-30:
After rebasing on the latest commits, only test_1() in demo_FEAbasic outputs undefined results: -1#INF

tasora, 2015-04-30:
The problem is fixed: the fact is that a 1D spring between two xyz nodes is a rank-deficient static problem (underdetermined: it would require at least two other springs). 
For the moment I fixed it by adding  0.01 terms on the diagonal by setting non-zero mass for the two nodes. 
Note: 
- the fact that atomic masses of nodes are added to K in the static analysis is a drawback of the current matrix-less bookkeeping code. It should be fixed in future otherwise one has to remember to set node atomic masses to zero before starting the static solver. 
- in future, the atomic node masses should be scaled as the element M mass matrix when doing KRMmatricesLoad() . Currently there is no straightforward way to do this, given the way that 'atomic' masses are handled in the code.

closed 2015-04-30 by tasora

jcmadsen, 2015-04-30:
Just re-ran the test 1-5 and I can confirm they are all correct now, thanks Alessandro