projectchrono/chrono #65: chrono failed to build on MacOsX Yosemite 10.10.5 with Clang

vacary, 2015-08-26:
Hi,

Chrono tag 2.0.0 failed to build on MacOsX Yosemite 10.10.5 with Clang.

ohana:chrono acary$ /usr/bin/c++ --version
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.5.0
Thread model: posix
ohana:chrono acary$ make
[  0%] Building CXX object CMakeFiles/ChronoEngine.dir/collision/ChCModelBullet.cpp.o
In file included from /Users/acary/Softs/chrono/src/collision/ChCModelBullet.cpp:25:
In file included from /Users/acary/Softs/chrono/src/physics/ChSystem.h:50:
In file included from /Users/acary/Softs/chrono/src/physics/ChLinksAll.h:38:
In file included from /Users/acary/Softs/chrono/src/physics/ChLinkPneumaticActuator.h:34:
In file included from /Users/acary/Softs/chrono/src/pneumatica/assepneumatico.h:20:
In file included from /Users/acary/Softs/chrono/src/pneumatica/sis_attuatore_3_2.h:22:
In file included from /Users/acary/Softs/chrono/src/pneumatica/pistone_3_2.h:22:
In file included from /Users/acary/Softs/chrono/src/pneumatica/pistone.h:23:
/Users/acary/Softs/chrono/src/pneumatica/volume.h:42:40: warning: expression result unused [-Wunused-value]
        volume() {rho=1.225; p0=101325, n = 1,4;g=p=v=v1=0;};
                                              ^
In file included from /Users/acary/Softs/chrono/src/collision/ChCModelBullet.cpp:32:
In file included from /Users/acary/Softs/chrono/src/collision/ChCConvexDecomposition.h:32:
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:309:53: error: friend declaration specifying a default argument must be a definition
        friend HACD * const                         CreateHACD(HeapManager * heapManager = 0);
                                                    ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:312:25: error: friend declaration specifying a default argument must be the only declaration
    inline HACD * const CreateHACD(HeapManager * heapManager) 
                        ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:309:53: note: previous declaration is here
        friend HACD * const                         CreateHACD(HeapManager * heapManager = 0);
                                                    ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:314:20: error: calling a private constructor of class 'HACD::HACD'
        return new HACD(heapManager);
                   ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:226:14: note: declared private here
                                                                                                        HACD(HeapManager * heapManager = 0);
                                                                                                        ^
1 warning and 3 errors generated.
make[2]: *** [CMakeFiles/ChronoEngine.dir/collision/ChCModelBullet.cpp.o] Error 1
make[1]: *** [CMakeFiles/ChronoEngine.dir/all] Error 2
make: *** [all] Error 2

hmazhar, 2015-08-27:
Hi Vincent,

Clang hasn’t been tested with chrono, I usually use OSX home-brew and "brew install gcc”
The gcc compilers work on OSX, if clang is a requirement please let me know and I can work on fixing the problems. 

Thanks,
Hammad

vacary, 2015-08-28:
Hi Hammad,

Ok, I'll switch to gcc for a while and try to fix the issue with clang later.

Thanks,
Vincent.

andrewseidl, 2015-08-28:
Try building the latest from the develop branch. The code in tag 2.0.0 is almost two years old at this point.

vacary, 2015-08-28:
Thanks for the advice.

Unfortunately, I got the same error. I will switch to gcc

ohana:chrono acary$ make VERBOSE=1
"/Applications/CMake 2.8-11.app/Contents/bin/cmake" -H/Users/acary/Softs/chrono/src -B/scratch/Vincent/Build/chrono --check-build-system CMakeFiles/Makefile.cmake 0
"/Applications/CMake 2.8-11.app/Contents/bin/cmake" -E cmake_progress_start /scratch/Vincent/Build/chrono/CMakeFiles /scratch/Vincent/Build/chrono/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/ChronoEngine.dir/build.make CMakeFiles/ChronoEngine.dir/depend
cd /scratch/Vincent/Build/chrono && "/Applications/CMake 2.8-11.app/Contents/bin/cmake" -E cmake_depends "Unix Makefiles" /Users/acary/Softs/chrono/src /Users/acary/Softs/chrono/src /scratch/Vincent/Build/chrono /scratch/Vincent/Build/chrono /scratch/Vincent/Build/chrono/CMakeFiles/ChronoEngine.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/ChronoEngine.dir/build.make CMakeFiles/ChronoEngine.dir/build
"/Applications/CMake 2.8-11.app/Contents/bin/cmake" -E cmake_progress_report /scratch/Vincent/Build/chrono/CMakeFiles 
[  0%] Building CXX object CMakeFiles/ChronoEngine.dir/collision/ChCModelBullet.cpp.o
/usr/bin/c++   -DBP_USE_FIXEDPOINT_INT_32 -DCH_API_COMPILE -DChronoEngine_EXPORTS -fPIC -I/Users/acary/Softs/chrono/src -I/Users/acary/Softs/chrono/src/collision/bullet -I/Users/acary/Softs/chrono/src/collision/gimpact -I/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD    -DAPPLE -D__APPLE__ -fpermissive -o CMakeFiles/ChronoEngine.dir/collision/ChCModelBullet.cpp.o -c /Users/acary/Softs/chrono/src/collision/ChCModelBullet.cpp
In file included from /Users/acary/Softs/chrono/src/collision/ChCModelBullet.cpp:25:
In file included from /Users/acary/Softs/chrono/src/physics/ChSystem.h:50:
In file included from /Users/acary/Softs/chrono/src/physics/ChLinksAll.h:38:
In file included from /Users/acary/Softs/chrono/src/physics/ChLinkPneumaticActuator.h:34:
In file included from /Users/acary/Softs/chrono/src/pneumatica/assepneumatico.h:20:
In file included from /Users/acary/Softs/chrono/src/pneumatica/sis_attuatore_3_2.h:22:
In file included from /Users/acary/Softs/chrono/src/pneumatica/pistone_3_2.h:22:
In file included from /Users/acary/Softs/chrono/src/pneumatica/pistone.h:23:
/Users/acary/Softs/chrono/src/pneumatica/volume.h:42:40: warning: expression result unused [-Wunused-value]
        volume() {rho=1.225; p0=101325, n = 1,4;g=p=v=v1=0;};
                                              ^
In file included from /Users/acary/Softs/chrono/src/collision/ChCModelBullet.cpp:32:
In file included from /Users/acary/Softs/chrono/src/collision/ChCConvexDecomposition.h:32:
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:309:53: error: friend declaration
      specifying a default argument must be a definition
        friend HACD * const                         CreateHACD(HeapManager * heapManager = 0);
                                                    ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:312:25: error: friend declaration
      specifying a default argument must be the only declaration
    inline HACD * const CreateHACD(HeapManager * heapManager) 
                        ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:309:53: note: previous declaration
      is here
        friend HACD * const                         CreateHACD(HeapManager * heapManager = 0);
                                                    ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:314:20: error: calling a private
      constructor of class 'HACD::HACD'
        return new HACD(heapManager);
                   ^
/Users/acary/Softs/chrono/src/collision/convexdecomposition/HACD/hacdHACD.h:226:14: note: declared private
      here
  ...HACD(HeapManager * heapManager = 0);
     ^
1 warning and 3 errors generated.
make[2]: *** [CMakeFiles/ChronoEngine.dir/collision/ChCModelBullet.cpp.o] Error 1
make[1]: *** [CMakeFiles/ChronoEngine.dir/all] Error 2
make: *** [all] Error 2

andrewseidl, 2015-08-28:
It still looks like you're on an out of date version of Chrono: some of the compiler options in there were removed/changed about a year ago, and the sources have been reorganized in the last couple of weeks.

`git checkout develop && git pull` and check the date of the most recent commit via `git log -1`.

It builds fine for me with Clang on 10.10.5, no GCC.

vacary, 2015-08-28:
This is fixed. Thank you very much and sorry for the noise. 
I was on the master branch...
Cheers

closed 2017-03-02 by thepianoboy