projectchrono/chrono #68: Chrono won't build with python enabled...

Jarmat, 2015-09-10:
I've had Chrono Engine up and running with Irrlicht, but want to add Python now to utilize the Solidworks plugin, but I can't get Chrono to build with Python.  For the record I'm using python3.4 and swig 3.0.7 (also tried older releases of python and swig but get the same errors).  During the build, first I get a lot of warnings that say:

/home/james/chrono/src/chrono_python/../chrono/core/ChSmartpointers.h:330: Warning 503: Can't wrap 'operator bool_type' unless renamed to a valid identifier.

But the make continues, then finally stops with two errors given below.  Any help would be greatly appreciated! 

[ 87%] Building CXX object src/chrono_python/CMakeFiles/_ChronoEngine_python_core.dir/__/__/python/ChModuleCorePYTHON_wrap.cxx.o
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx: In member function ‘virtual void chrono::ChLogPython::Output(const char*, size_t)’:
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:6890:29: warning: format not a string literal and no format arguments [-Wformat-security]
     PySys_WriteStdout(buffer);
                             ^
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx: In function ‘PyObject* PyInit__ChronoEngine_python_core()’:
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:531125:133: error: taking address of temporary [-fpermissive]
   SWIG_Python_SetConstant(d, "CSYSNULL",SWIG_NewPointerObj(SWIG_as_voidptr(&chrono::ChCoordsys< double >(chrono::VNULL,chrono::QNULL)),SWIGTYPE_p_chrono__ChCoordsysT_double_t, 0 ));
                                                                                                                                     ^
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:1172:89: note: in definition of macro ‘SWIG_NewPointerObj’
 #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
                                                                                         ^
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:531125:60: note: in expansion of macro ‘SWIG_as_voidptr’
   SWIG_Python_SetConstant(d, "CSYSNULL",SWIG_NewPointerObj(SWIG_as_voidptr(&chrono::ChCoordsys< double >(chrono::VNULL,chrono::QNULL)),SWIGTYPE_p_chrono__ChCoordsysT_double_t, 0 ));
                                                            ^
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:531126:133: error: taking address of temporary [-fpermissive]
   SWIG_Python_SetConstant(d, "CSYSNORM",SWIG_NewPointerObj(SWIG_as_voidptr(&chrono::ChCoordsys< double >(chrono::VNULL,chrono::QUNIT)),SWIGTYPE_p_chrono__ChCoordsysT_double_t, 0 ));
                                                                                                                                     ^
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:1172:89: note: in definition of macro ‘SWIG_NewPointerObj’
 #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
                                                                                         ^
/home/james/chrono/chrono_build/python/ChModuleCorePYTHON_wrap.cxx:531126:60: note: in expansion of macro ‘SWIG_as_voidptr’
   SWIG_Python_SetConstant(d, "CSYSNORM",SWIG_NewPointerObj(SWIG_as_voidptr(&chrono::ChCoordsys< double >(chrono::VNULL,chrono::QUNIT)),SWIGTYPE_p_chrono__ChCoordsysT_double_t, 0 ));
                                                            ^
make[2]: *** [src/chrono_python/CMakeFiles/_ChronoEngine_python_core.dir/__/__/python/ChModuleCorePYTHON_wrap.cxx.o] Error 1
make[1]: *** [src/chrono_python/CMakeFiles/_ChronoEngine_python_core.dir/all] Error 2
make: *** [all] Error 2

cekees, 2016-05-26:
Any update on the Python interface? Has this issue been solved  in the master branch?

tasora, 2016-05-26:
> Bug from

Dear Chris,

  the master branch compatibility with Python is not maintained anymore.
Please use the 'develop' branch of Chrono.

It compiles ok with Python 3.2 on my system.

Alessandro

cekees, 2016-05-26:
Ah, OK. I think I remember that now.  I think the issue was that it wouldn't compile in Python 2.7.x.  do you know if there will be (or is) a backport to some 2.x Python?

closed 2016-10-19 by tasora