projectchrono/chrono #79: Includes don't work correctly when using make install to install to /usr/local/.

dylanede, 2016-03-10:
On Unix-like systems at least, installing to `/usr/local/` so that the chrono include directory is found on the default include search path causes the problem that although headers can be found using `#include "chrono/some_header.h"`, any chrono headers included by that header assume a root of chrono, not its containing directory (`/usr/local/include/` in this case), so they're not found. As an example, `chrono/physics/ChSystem.h` includes `core/ChLog.h`, when really it should include `chrono/core/ChLog.h`. Could this be fixed at all? Is there a reason that it's like this?

elliotcm, 2016-12-24:
@dylanede It's been a while since you asked this so probably not relevant, but I've found similar issues with this, my workaround was to add `-I /usr/local/include/chrono` to my compile flags so the shorter paths match.  Not the best but functional.

elliotcm, 2016-12-24:
Hi @rserban, thanks for this.  The project I'm building is very simple and just uses Make with no IDE, so I'm leaving CMake aside for now.

If it helps anyone else these are my current compile flags which happily compile the Physical System example:
```
CFLAGS := -g -Wall -std=c++11 -mavx
LIB := -L lib -lchronoengine
INC := -I include -I /usr/local/include/chrono
```

rserban, 2016-12-26:
@elliotcm, I just pushed a set of changes throughout the entire project that should take care of this.  When you get a chance, please check that you are now able to compile without `-I /usr/local/include/chrono`

elliotcm, 2016-12-26:
@rserban 👍 worked fine for `#include "chrono/physics/ChSystem.h"`

closed 2016-12-30 by rserban

huckl3b3rry87, 2017-09-08:
@rserban I am just getting started with Chrono and running into similar issues while trying to find these header files on a Linux system. 

Here is my cmake setup:

[image]

Which produces:
```
febbo@febbo-HP-ZBook-17-G2:~/Documents/workspace/tutorials/chrono_tutorials/build$ ninja
[1/2] Building CXX object demo_Steerin....dir/demo_VEH_SteeringController.cpp.o
FAILED: /usr/bin/c++      -MMD -MT demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o -MF demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o.d -o demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o -c /home/febbo/Documents/workspace/tutorials/chrono_tutorials/source/demo_SteeringController/demo_VEH_SteeringController.cpp
/home/febbo/Documents/workspace/tutorials/chrono_tutorials/source/demo_SteeringController/demo_VEH_SteeringController.cpp:22:37: fatal error: chrono/core/ChFileutils.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
```
In previous attempts, I tried copying the header files from ``opt/chrono/chrono_source/src`` to ``/usr/local/include``, and this solved the issue for the file that I was trying to find, but then, inside the file that I was trying to find I get similar path issues such as:
```
febbo@febbo-HP-ZBook-17-G2:~/Documents/workspace/tutorials/chrono_tutorials/build$ ninja
[1/2] Building CXX object demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o
FAILED: /usr/bin/c++      -MMD -MT demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o -MF demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o.d -o demo_SteeringController/CMakeFiles/demo_SteeringController.dir/demo_VEH_SteeringController.cpp.o -c /home/febbo/Documents/workspace/tutorials/chrono_tutorials/source/demo_SteeringController/demo_VEH_SteeringController.cpp
In file included from /usr/local/include/chrono/core/ChFileutils.h:16:0,
                 from /home/febbo/Documents/workspace/tutorials/chrono_tutorials/source/demo_SteeringController/demo_VEH_SteeringController.cpp:22:
/usr/local/include/chrono/core/ChApiCE.h:18:30: fatal error: chrono/ChVersion.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
```
Since @rserban said he fixed this issue more than a year ago, I am assuming that I am doing something silly, but I cannot figure it out.

Any help or advice would be greatly appreciated. Thanks!

huckl3b3rry87, 2017-09-11:
After using the example https://github.com/projectchrono/chrono/blob/develop/template_project/my_example.cpp I was able to get things working.