projectchrono/chrono #91: Chrono::Parallel OSX installation

markrans, 2016-07-13:
Hi can we please get some documentation on setting up Chrono with OSX? I've managed to get Chrono::Engine and Irrlicht working but have become stuck with Parallel. I've tried using GCC instead of Clang and have attempted it through XCode and Unix makefiles but the compilation always crashes (the current problematic file seems to be real4.h). 
Many thanks, Mark

hmazhar, 2016-07-14:
Hi Mark,
Can you provide the following?
Full error message
Version of GCC
Version of OSX

markrans, 2016-07-14:
Hi Hammad, I'm using OSX 10.11.5 and GCC 5.4.0 hooked into XCode 7.3.1 using your plugin (though CMake's log says it's still using AppleClang 7.3.0.7030031 yet the CMAKE_CX_COMPILER and CMAKE_C_COMPILER fields are set to /.../g++-5 and /.../gcc-5. I set CMAKE_AR, CMAKE_LINKER, CMAKE_NM and CMAKE_RANLIB to what seemed like appropriate execs for GCC5). 
The error message is:

Ld build/lib/Debug/libChronoEngine.dylib normal x86_64
    cd /.../chrono-develop
    export MACOSX_DEPLOYMENT_TARGET=10.11
    /usr/local/Cellar/gcc5/5.4.0/bin/g++-5 -arch x86_64 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -L/.../chrono-develop/build/lib/Debug -F/.../chrono-develop/build/lib/Debug -filelist /Users/mark1/Documents/Chrono/chrono-develop/build/src/chrono/Chrono.build/Debug/ChronoEngine.build/Objects-normal/x86_64/ChronoEngine.LinkFileList -install_name /.../chrono-develop/build/lib/Debug/libChronoEngine.dylib -mmacosx-version-min=10.11 -Xlinker -no_deduplicate -dynamiclib -Wl,-headerpad_max_install_names -lpthread -single_module -Xlinker -dependency_info -Xlinker /.../chrono-develop/build/src/chrono/Chrono.build/Debug/ChronoEngine.build/Objects-normal/x86_64/ChronoEngine_dependency_info.dat -o /.../chrono-develop/build/lib/Debug/libChronoEngine.dylib

Undefined symbols for architecture x86_64:
  "std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::compare(char const*) const", referenced from:
      chrono::ChClassRegisterABSTRACT<chrono::ChAsset>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChAsset.o
      chrono::ChClassRegisterABSTRACT<chrono::ChAssetLevel>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChAssetLevel.o
      chrono::ChClassRegisterABSTRACT<chrono::ChBoxShape>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChBoxShape.o
      chrono::ChClassRegisterABSTRACT<chrono::ChCamera>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChCamera.o
      chrono::ChClassRegisterABSTRACT<chrono::ChCapsuleShape>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChCapsuleShape.o
      chrono::ChClassRegisterABSTRACT<chrono::ChColor>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChColor.o
      chrono::ChClassRegisterABSTRACT<chrono::ChColorAsset>::get_conventional_name(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&) in ChColorAsset.o

Followed by 199 similar errors. Any ideas? Thanks

hmazhar, 2016-07-15:
Would it be possible to try using makefiles directly?, We don't support using XCode on OSX currently.

thepianoboy, 2017-03-06:
Chrono parallel builds on macOS Sierra using the stock version of Apple clang as of commit db1ad88 (at least). GNU Make, Irrlicht, and boost are available via brew and should work "out of box". Blaze versions up to 2.6 are confirmed to work on macOS. You shouldn't require any special flags for CMake, but setting USE_PARALLEL_CUDA to OFF seems to be a good idea on Unix platforms at this time. 

Proper documentation will be available at projectchrono.org in the near future.

thepianoboy, 2019-05-17:
Feel free to take a look at http://buildbot.projectchrono.org/#/builders/7 for example configurations on macOS. Closing stale issue.

closed 2019-05-17 by thepianoboy