projectchrono/chrono #93: Irrlicht shared library is installed to the wrong place

tapir, 2016-08-03:
I'm on arch linux and I enabled Irrlicht module. Everything compiled successfully but installation failed with this error: 

```
Error copying file "/home/tapir/Documents/chronoengine-git/src/chrono/build/lib/libChronoEngine_irrlicht.so" to "/usr/bin/DLLs/.".
```

I checked and indeed libChronoEngine_irrlicht.so is there. So I'm guessing /usr/bin/DLLs throws the error.

tapir, 2016-08-06:
Any pointers how I can compile the python support?

cekees, 2016-11-18:
@rserban I'm interested in some pointers for compiling with python support as well, especially under Python 2.7

sanguinariojoe, 2016-12-23:
I fixed that replacing at src/chrono_python/CMakeLists.txt (Line 101) the following lines:

```
SET(CH_PYTHONDIR_PYD "${CH_PYTHONDIR}/DLLs")
SET(CH_PYTHONDIR_PY  "${CH_PYTHONDIR}/Lib")
```
by:

```
IF(${CMAKE_SYSTEM_NAME} MATCHES "Linux")
    SET(CH_PYTHONDIR_PYD "${PROJECT_BINARY_DIR}/python")
    SET(CH_PYTHONDIR_PY  "${PROJECT_BINARY_DIR}/python")
ELSE(${CMAKE_SYSTEM_NAME} MATCHES "Linux")
    SET(CH_PYTHONDIR_PYD "${CH_PYTHONDIR}/DLLs")
    SET(CH_PYTHONDIR_PY  "${CH_PYTHONDIR}/Lib")
ENDIF()
```

After that, you may compile without errors, and get all the python stuff at `python` subfolder (respect the place where you are compiling on)

That's a dirty trick, but good enough for the moment. Maybe @tapir would consider integrating it, or something like (MAC is not even considered bu the patch)...

rserban, 2016-12-30:
I just pushed (to the `develop` branch) a rewrite of the CMakeLists.txt script for the Chrono::Python module (and other associated code fixes).   This should fix issues with configuring and building Python support, both on Windows and Linux platforms.  

I tested this on the following:
- Windows 10, Visual Studio 2013 Ultimate, SWIG 3.0.8, Python 3.6.0
- Windows 8, Visual Studio 2013 Pro, SWIG 3.0.8, Python 3.3
- Ubuntu 14.04 LTS, gcc 4.9.4, SWIG 3.0.10, Python 2.7.6
- CentOS Linux rel. 7.2.1511, SWIG 3.0.7, Python 3.6.0

Note that, after build (and optionally, installation), you must set the PYTHONPATH environment variable to point to the generated modules in either the BUILD or the INSTALL tree (during CMake configuration, the script will display the appropriate directories to be added to PYTHONPATH).  See the [installation guide](http://api.chrono.projectchrono.org/module_python_installation.html) for more details.

@cekees: As you can see, I also tested this with Python 2.7 and all seemed to work fine (without changes to any of the Python demo scripts -- which is a bit odd, but maybe it has to do with using a more recent SWIG version).  Please give it a try and let me know how it goes for you.

closed 2016-12-30 by rserban