projectchrono/chrono #106: CHRONO install problem on linux

yunxiaoshan, 2017-02-27:
Hi, I am trying to install on linux,(ubuntu 14.04). But I get the following error.
```
CMake Error at build/CMakeFiles/git-data/grabRef.cmake:36 (file):
  file failed to open for reading (No such file or directory):

    /home/yunxiao/Downloads/chrono-develop/build/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  cmake/GetGitRevisionDescription.cmake:77 (include)
  CMakeLists.txt:76 (get_git_head_revision)

```
what may be the problem?
-shan

thepianoboy, 2017-03-03:
Could you give us a bit more information? Your CMake version would be a good start, and it wouldn't hurt to have the options you've passed to CMake so far.

yunxiaoshan, 2017-03-07:
cmake v3.2.2, I do not pass any options to CMake.

thepianoboy, 2019-05-17:
Closing as stale. This issue does not seem to be reproducible on the most recent release of Chrono.

closed 2019-05-17 by thepianoboy