projectchrono/chrono #109: Using RUNGEKUTTA45 timestepper breaks IRR_emit_processor.cpp demo.

kkimdev, 2017-05-03:
I'm not sure if it's a bug or not, but if I put `mphysicalSystem.SetTimestepperType(ChTimestepper::Type::RUNGEKUTTA45);` in `demo_IRR_emit_processor.cpp`, then the objects spawned doesn't collide the ground.

tasora, 2017-05-03:
Dear Kim,
The RUNGEKUTTA45 timestepper does not support contacts. At the moment the only timestepper that supports DVI contacts is the default one.

Alessandro

Da: Kibeom Kim
Inviato: Mercoledì 3 Maggio 07:54 AM
Oggetto: [projectchrono/chrono] Using RUNGEKUTTA45 timestepper breaks IRR_emit_processor.cpp demo. (#109)
A: projectchrono/chrono
Cc: Subscribed

I'm not sure if it's a bug or not, but if I put mphysicalSystem.SetTimestepperType(ChTimestepper::Type::RUNGEKUTTA45); in demo_IRR_emit_processor.cpp, then the objects spawned doesn't collide the ground.

—

Il 5xmille a favore dell’Università di Parma (indicando il codice fiscale 00308780345) è un gesto tanto semplice quanto importante per creare nuove opportunità di lavoro per giovani di talento nell'ambito della ricerca scientifica.

kkimdev, 2017-05-03:
Oh I see, thanks!

closed 2017-05-03 by kkimdev