projectchrono/chrono #110: Meta: contributor agreement

storrgie, 2017-05-03:
This came up in conversation and I figure the best thing to do is raise it here for consideration. I think there are quite a few people from within the government working on aspects of chrono that should be pushed back into the upstream. Chrono is already licensed so there is way less complexity from a government participation standpoint. There is a sticking point on contributions from people who cannot assign copyright, [ARL is hashing some of these ideas out over here](https://github.com/USArmyResearchLab/ARL-Open-Source-Guidance-and-Instructions).

I am not a lawyer, but I think that for this project a simple contributor agreement that reassigns copyright to the project maintainers would get the government people around any worry that their legal department would get obnoxious about them attempting to contribute upstream. A contributor agreement that has to be agreed to by the person submitting the pull request / commit would get around the worry that their work would be public domain and otherwise not encapsulated by the projects license. 

Mainly, its hard to get lawyers to agree to let us contribute code upstream. If the choices were all made for us by the upstream source it will reduce the discussion space.

rserban, 2017-05-04:
Andrew,

It goes without saying that we would be thrilled to get back such contributions and incorporate them in the code base.  This would be beneficial for all parties: on one hand we enrich Chrono for all users; on the other hand, we would automatically care of maintaining, keeping in sync, and updating any such new contributions as the Chrono code evolves (and they would be part of the automated builds and testing we have set up).

As you suggested, assigning the copyright to us (the project maintainers) would be the simplest way.  Otherwise, it gets complicated to have to deal with multiple licenses and copyright notices (less so for more "stand-alone" modules, but certainly for contributions affecting the existing code base).

I looked briefly over the  ARL open-source guidelines you linked to.  Correct me if I'm wrong, but that seems to deal mostly with software originating at ARL and contributions from outside - practically the opposite of what we're talking about here.   Is ARL looking into this other scenario as well?

Of course, we would need to talk to legal at the university and run a template CLA by them.

--Radu

storrgie, 2017-05-05:
Right, ARL's repo is, I think, more for the established governmental project that has been taken from inside to outside and licensed with the nuance of copyright not being assignable to the government employee. I linked it to display that there is both willingness on the governments side, as well as nuance that is complex to deal with when the government establishes a project.

In the case of Chrono this would be simpler because you could just place a contributions statement that would reassign the copyright to the project. However IANL, so it would be very good to have your counsel consider the ideas and find some sort of precedence to follow.

If you could get a contributions statement in place, it would clear up some of the discussions I've heard about whether internal work could be contributed upstream. It would be, internally, much more clear about what an upstream contribution would mean (our lawyers wouldn't have as much design space to consider when we're asking them to allow us to issue a pull request).