projectchrono/chrono #111: error C2259 when build template_project with Visual Studio 2015 update 3

wgm096350, 2017-05-28:
I try to build the project in folder template_project. There are only 2 file in this folder: CMakeLists.txt and my_example.cpp. While buiding, I follow the guidelines from  [http://api.chrono.projectchrono.org/tutorial_install_project.html](url). CMAKE can generate *.sln without any problem. But when compiling the project in VS 2015, it shows "error C2259 cannot instantiate abstract class due to following members". The error is from this line:
[image]

What should I do to resolve this issue?

tasora, 2017-05-28:
hi

recently we did a refactoring, and ChSystem now is called ChSystemNSC , where NSC means Non Smooth Contact.

Please change ChSystem into ChSystemNSC.

regards,

Alessandro Tasora

Il 28/05/2017 14:31, wgm096350 ha scritto:

I try to build the project in folder template_project. There are only 2 file in this folder: CMakeLists.txt and my_example.cpp. While buiding, I follow the guidelines from http://api.chrono.projectchrono.org/tutorial_install_project.html<url>. CMAKE can generate *.sln without any problem. But when compiling the project in VS 2015, it shows "error C2259 cannot instantiate abstract class due to following members". The error is from this line:
[qq 20170528202951]<https://cloud.githubusercontent.com/assets/24369346/26528756/8847fd1c-43e4-11e7-9961-693ad560c55a.png>

What should I do to resolve this issue?

—

Il 5xmille a favore dell’Università di Parma (indicando il codice fiscale 00308780345) è un gesto tanto semplice quanto importante per creare nuove opportunità di lavoro per giovani di talento nell'ambito della ricerca scientifica.

rserban, 2017-05-28:
To add to Alessandro's answer, ChSystem still exists but it is now an abstract base class (hence the compilation error you were getting).   Note that you'll also have to include the header 'ChSystemNSC.h'.  
I pushed the fix in the 'develop' branch.
--Radu

closed 2017-05-28 by rserban

wgm096350, 2017-05-29:
Thanks for the tips！ It works.