projectchrono/chrono #113: error: '__m256d'

cekees, 2017-07-18:
@Milad-Rakhsha we are not able to build Chrono projects on ubuntu 14.04 because of the following compil errors. I Seem to recall that we fixed this, but maybe I'm wrong. Here's a more complete error trace:

```
mpicc -fno-strict-aliasing -g -O2 -g -fwrapv -w -fPIC -I/buildbot/Ubuntu_14_04/build/linux2/lib/python2.7/site-packages/numpy/core/include -Iproteus -I/buildbot/Ubuntu_14_04/build/linux2/include -I/buildbot/Ubuntu_14_04/build/linux2/include/chrono/collision/bullet -I/buildbot/Ubuntu_14_04/build/linux2/include/python2.7 -c proteus/mbd/pyChronoCore.cpp -o build/temp.linux-x86_64-2.7/proteus/mbd/pyChronoCore.o -std=c++11
3651In file included from /buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrixNM.h:18:0,
3652                 from /buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix33.h:18,
3653                 from /buildbot/Ubuntu_14_04/build/linux2/include/chrono/collision/ChCCollisionModel.h:20,
3654                 from /buildbot/Ubuntu_14_04/build/linux2/include/chrono/collision/ChCCollisionInfo.h:17,
3655                 from /buildbot/Ubuntu_14_04/build/linux2/include/chrono/collision/ChCCollisionSystem.h:18,
3656                 from /buildbot/Ubuntu_14_04/build/linux2/include/chrono/physics/ChSystem.h:26,
3657                 from proteus/mbd/ChMoorings.h:8,
3658                 from proteus/mbd/pyChronoCore.cpp:458:
3659/buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix.h: In member function 'void chrono::ChMatrix<Real>::MatrMultiplyAVX(const chrono::ChMatrix<double>&, const chrono::ChMatrix<double>&)':
3660/buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix.h:547:17: error: '__m256d' was not declared in this scope
3661                 __m256d sum = _mm256_setzero_pd();
3662                 ^
3663/buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix.h:547:25: error: expected ';' before 'sum'
3664                 __m256d sum = _mm256_setzero_pd();
3665                         ^
3666/buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix.h:549:29: error: expected ';' before 'ymmA'
3667                     __m256d ymmA = _mm256_broadcast_sd(A_add + A_NCol * rowA + elem);
3668                             ^
3669/buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix.h:550:29: error: expected ';' before 'ymmB'
3670                     __m256d ymmB = _mm256_loadu_pd(B_add + elem * B_NCol + colB);
3671                             ^
3672/buildbot/Ubuntu_14_04/build/linux2/include/chrono/core/ChMatrix.h:551:29: error: expected ';' before 'prod'
3673                     __m256d prod = _mm256_mul_pd(ymmA, ymmB);```

Milad-Rakhsha, 2017-07-18:
I am aware of this issue and I believe that this has to do with the CMake configuration for AVX support. So, at the configuration stage in CMAKE it is found that your system does support AVX. Consequently, CHRONO_HAS_AVX is defined in the build/chrono/ChConfig.h, which is included in ChMatrix.h. This allows for vectorized matrix multiplications. However, it seems like your system does not find the immintrin.h for some reason at the build stage. Could you see if this system header file is available in your compiler's include? I think a work around is to add the path of the immintrin.h to your CPATH. 
By the way what commit produces this error?

drcoopa, 2021-09-10:
Hi!  I'm having a similar problem of being unable to build Chrono projects and not able to resolve it.  I just installed Chrono on my Mac (Big Sur).   

Same errors about expected characters.  For example:

                                                                              
/usr/local/include/chrono/core/ChMatrix.h:129:93: error: expected '>'
/usr/local/include/chrono/core/ChMatrix.h:129:48: note: to match this '<'
using ChRowVectorRef = Eigen::Ref<Eigen::Matrix<double, 1, Eigen::Dynamic, Eigen::RowMajor>>;
                                               ^
/usr/local/include/chrono/core/ChMatrix.h:129:93: error: expected a type
using ChRowVectorRef = Eigen::Ref<Eigen::Matrix<double, 1, Eigen::Dynamic, Eigen::RowMajor>>;
                                                                                            ^
/usr/local/include/chrono/core/ChMatrix.h:129:93: error: expected '>'
/usr/local/include/chrono/core/ChMatrix.h:129:34: note: to match this '<'

I tried to point to immintrin.h with a symbolic link within the include folder, but that didn't seem to help.

rserban, 2021-10-14:
Sorry for the late reply.  Did you resolve this issue?
If not, could please try to disable SIMD support during CMake configuration (uncheck `USE_SIMD`)?

closed 2025-11-28 by rserban