projectchrono/chrono #114: compiling in ubuntu 16.04 problem

kmolLin, 2017-07-24:
[image]

My system is ubuntu 16.04 want to compile with python3.5  The cmakeList is below

[image]

[image]

and the irrlicht I have compile two like this 

[image]

Did I have wrong setting in linux 
Best regard

kmolLin, 2017-07-25:
Oh I'm sorry I find the error on my computer ,because I have setup the swig form error excutable . So that there is no swig for the compiler to compile, But there have a new question on compile with swig, I use the swig(3.0.12).
[image]

uebian, 2026-01-12:
We no longer support the old OS and SWIG version. Please open a new ticket if the issue still persists with newer version of Irrlicht, SWIG and Ubuntu.

closed 2026-01-12 by uebian