projectchrono/chrono #115: support for linking to Gazebo

huckl3b3rry87, 2017-09-07:
I see that Chrono has some capabilities for sensor models, but from what I can tell they are not yet as mature as the ones in Gazebo. On the other had Gazebo has a limited vehicle dynamic support and Chrono seems to do a much better job with this (from what I can tell). I noticed the repo that connects Gazebo and Chrono (to get the best of both worlds) called Chrono_Gazebo, but this package is out of date and I started to update it with https://github.com/amelmquist/Chrono_Gazebo/pull/6, but I am not very familiar with these tools, so it is my feeling that this will take quite a bit of work for me to complete. I noticed that Chrono supports interfacing with other software packages such as MATLAB and SolidWorks, but there is nothing for Gazebo.

Is there a reason that the Chrono_Gazebo project is not maintained and an option to use in the current Chrono?

Is there any potential to update this repo and maintain it with new versions of ROS, Gazebo and Chrono?

Thanks!

DanNegrut, 2017-09-08:
Hi,
thanks for your email.
There are Chrono models that run one second in one day (FEA tires, running on one million particles).
At the other end of the spectrum there are HMMWV models that are simple and run in real time. 
Also there are models in between, both wheeled to tracked vehicles.
I guess it depends on the level of fidelity you need.
I would suggest dropping an email to Radu Serban serban@wisc.edu, he's working on the Chrono::Vechile side.
For autonomous vehicle simulation, drop me an email (negrut@wisc.edu) - we are pursuing this avenue at this time. Outside Gazebo.
Just curious, are you working in Professor Huei Peng's group?
Cheers,
Dan

jyomk, 2022-06-03:
Hello, I'm new to using Chrono and ROS but I was trying to understand if there is any documentation to simulate off road vehicles using Chrono physics engine and Gazebo sensor models? Thank you

DanNegrut, 2022-06-03:
No, unfortunately there is no such documentation.

From a high vantage point, you can do the following: drive the Chrono vehicle around in a Chrono simulation based on input determined by your ROS autonomy stack (steering wheel input, acceleration & brake pedal input). Then at a certain frequency you pick up the Chrono vehicle position/orientation/velocity and send back to Gazebo. There you can update the Gazebo state of the vehicle “ghost” based on the position/orientation/velocity you got from Chrono. You keep doing this over and over.

Alternatively, Chrono has its own suite of sensor models – camera, lidar, IMU, GPS. And it has a ROS autonomy stack that it works with – see here:
https://uwmadison.box.com/s/tb1iqv3f7dfa6kyklqcgao2v0kp4mdtu
This is work in progress.

There is documentation for this, see here:
http://projects.sbel.org/autonomy-toolkit/

Thanks for your interest.
Dan

-------------------------------------------------
Bernard A. and Frances M. Weideman Professor
NVIDIA CUDA Fellow
Director, Wisconsin Applied Computing Center
Department of Mechanical Engineering
Department of Computer Science
University of Wisconsin - Madison
4150ME, 1513 University Avenue
Madison, WI 53706-1572
608 772 0914
http://sbel.wisc.edu/
http://projectchrono.org/
-------------------------------------------------