projectchrono/chrono #118: fatal error: chrono/physics/ChContactContainerDEM.h: No such file or directory

olivertai, 2017-11-22:
Hi, I am trying to build the program SceneNetRGB-D, I have successfully installed chrono and irrlicht 1.8.2,
but when I try to compile SceneNet, I get problem like following:
`/mntExt/DW/codes/scenenetrgb-d/scene_generator/scenenet_room_generator.cpp:23:50: fatal error: chrono/physics/ChContactContainerDEM.h: No such file or directory
compilation terminated.
make[2]: *** [CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o] error 1
make[1]: *** [CMakeFiles/scenenet_room_generator.dir/all] error 2
make: *** [all] error 2
`
But I can't find a file container named "physics" in projectchrono, can someone help me?
Thank you.

rserban, 2017-11-22:
You do not mention what version of Chrono you have downloaded and installed.  But the subdirectory `physics` certainly exists ([see here](https://github.com/projectchrono/chrono/tree/develop/src/chrono/physics)).

Also, I'm not sure what version (commit) of Chrono you are supposed to link to when building SceneNetRGB, but it's clear that it's an older version.  Indeed, the class `ChContactContainerDEM` was renamed to `ChContactContainerSMC` a while ago (on May 2, 2017).  You may want to contact the maintainers of SceneNetRGB and ask them what version of Chrono you are supposed to use.

Radu

olivertai, 2017-11-22:
@rserban
Thank you for the advice. The project SceneNet needs 2.0.0 version of projectchrono. I downloaded the code successfully but I found that the installation tutorial page is missing.[http://www.projectchrono.org/mediawiki/index.php/Main_Page](url)
Could you tell me how to organize and compile the files since I can't find any cmakelist or makelist?
Thank you very much.

rserban, 2017-11-22:
I doubt strongly that SceneNetRGB-D links against version 2.0 of Chrono.  That is more than 4 years old.   I suspect that the SceneNet authors used a much more recent version.  The thing is that we did not have a tagged release of Chrono for a very long time.

Indeed, looking at the Bitbucket SceneNetRGB-D repository, they indicate a **particular** Chrono commit to use:  b5af3f50c39cf2f40e4b05b3115ebb23a69ec8c9.   What you need to do is checkout that version of Chrono.  This is only about 16 months old and you should be able to configure and build it using CMake.

Radu

closed 2017-12-04 by olivertai