projectchrono/chrono #119: el::base undefined references when using module parallel

olivertai, 2017-12-04:
Hello, I'm trying to turn my code a bit faster by adding the parallel module. I installed boost, blaze and CUDA, and then I enabled MODULE_PARALLEL in CMake. I recompiled project chrono with success, but when I tried to include files like "chrono_parallel/physics/ChSystemParallel.h" and "chrono_parallel/collision/ChCollisionModelParallel.h" and compile my project I got errors like:
`
[ 50%] Building CXX object CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o
[100%] Linking CXX executable scenenet_room_generator
CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o：in function ‘el::base::debug::crashReason(int)’中：
scenenet_room_generator.cpp:(.text+0x279)：对‘el::base::elStorage’ undefined reference
CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o：in function ‘el::base::LogFormat::updateFormatSpec()’中：
scenenet_room_generator.cpp:(.text._ZN2el4base9LogFormat16updateFormatSpecEv[_ZN2el4base9LogFormat16updateFormatSpecEv]+0x905)：对‘el::base::utils::s_currentUser’ undefined reference
scenenet_room_generator.cpp:(.text._ZN2el4base9LogFormat16updateFormatSpecEv[_ZN2el4base9LogFormat16updateFormatSpecEv]+0x944)：对‘el::base::utils::s_currentUser’ undefined reference
scenenet_room_generator.cpp:(.text._ZN2el4base9LogFormat16updateFormatSpecEv[_ZN2el4base9LogFormat16updateFormatSpecEv]+0x9c3)：对‘el::base::utils::s_currentHost’ undefined reference
CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o：in function ‘el::base::LogDispatcher::dispatch()’中：
scenenet_room_generator.cpp:(.text._ZN2el4base13LogDispatcher8dispatchEv[_ZN2el4base13LogDispatcher8dispatchEv]+0x52)：对‘el::base::elStorage’ undefined reference
scenenet_room_generator.cpp:(.text._ZN2el4base13LogDispatcher8dispatchEv[_ZN2el4base13LogDispatcher8dispatchEv]+0x91)：对‘el::base::elStorage’ undefined reference
scenenet_room_generator.cpp:(.text._ZN2el4base13LogDispatcher8dispatchEv[_ZN2el4base13LogDispatcher8dispatchEv]+0xac)：对‘el::base::elStorage’ undefined reference
scenenet_room_generator.cpp:(.text._ZN2el4base13LogDispatcher8dispatchEv[_ZN2el4base13LogDispatcher8dispatchEv]+0xfb)：对‘el::base::elStorage’ undefined reference
CMakeFiles/scenenet_room_generator.dir/scenenet_room_generator.cpp.o：in function ‘el::base::MessageBuilder::initialize(el::Logger*)’中：
scenenet_room_generator.cpp:(.text._ZN2el4base14MessageBuilder10initializeEPNS_6LoggerE[_ZN2el4base14MessageBuilder10initializeEPNS_6LoggerE]+0x1c)：对‘el::base::elStorage’ undefined reference
`
Could you help me with that? Thank you.

rserban, 2017-12-05:
Are you by any chance making calls to some Easylogging functions in your project (besides those from within Chrono::Parallel itself)?   If so, you may have to initialize it by defining `INITIALIZE_EASYLOGGINGPP`.  See [this GitHub issue](https://github.com/muflihun/easyloggingpp/issues/535).   

For use within Chrono::Parallel, this is done in the file ChSystemParallel.cpp (line 46 in the current version in the 'develop' branch).

Radu

olivertai, 2017-12-06:
@rserban 
Problem solved. thx.

closed 2017-12-06 by olivertai

rserban, 2017-12-06:
@olivertai Could you please mention what the problem ended up being?  Was it the missing initialization of easylogging in user code?