projectchrono/chrono #122: Build against external version of bullet

bholcomb, 2018-03-05:
I'm looking to build chrono against the version of bullet that is currently integrated into my project (currently version 2.82).   Are there any modifications to the included version of bullet that might prevent me from building against the version I already have integrated?

Cheers, 

Bob

rserban, 2018-03-21:
Hi Bob,

Yes, the Bullet version included in Chrono (for collision detection) is pretty significantly modified.  This would not be a trivial task.
Note also that you will not be able to link a project to both Chrono and Bullet.  
We are aware of all these issues and plan on addressing them, but that's unlikely to happen very soon.

--Radu

tasora, 2018-03-21:
Dear Bob,

  the Bullet code embedded in Chrono has some modifications, you cannot 
just switch to the official Bullet and expect that all will work as 
fine. If you want to have an idea of the part that I modified, try 
searching for

***ALE

in the c++ sources of Bullet because I added some comments with this 
tag, to remember where I changed the original Bullet source.

regards.

Alessandro Tasora

closed 2022-02-02 by rserban