projectchrono/chrono #124: build failure on gcc8.1

bartoszek, 2018-06-13:
Have some minor isse when building against new gcc (8.1)
```
[ 96%] Built target demo_PAR_ballsSMC
Scanning dependencies of target demo_GL_cohesion
In file included from /opt/cuda/include/thrust/detail/reference.h:173,
                 from /opt/cuda/include/thrust/memory.h:25,
                 from /opt/cuda/include/thrust/device_ptr.h:25,
                 from /opt/cuda/include/thrust/device_malloc_allocator.h:25,
                 from /opt/cuda/include/thrust/device_vector.h:25,
                 from /home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiDataManager.cuh:24,
                 from /home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiInterface.h:22,
                 from /home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiInterface.cpp:18:
/opt/cuda/include/thrust/detail/reference.inl: In instantiation of ‘thrust::reference<Element, Pointer, Derived>::value_type thrust::reference<Element, Pointer, Derived>::strip_const_get_value(const System&) const [with System = thrust::cuda_cub::tag; Element = chrono::fsi::Real3; Pointer = thrust::device_ptr<chrono::fsi::Real3>; Derived = thrust::device_reference<chrono::fsi::Real3>; thrust::reference<Element, Pointer, Derived>::value_type = chrono::fsi::Real3]’:
/opt/cuda/include/thrust/detail/reference.inl:105:54:   required from ‘thrust::reference<Element, Pointer, Derived>::value_type thrust::reference<Element, Pointer, Derived>::convert_to_value_type(System*) const [with System = thrust::cuda_cub::tag; Element = chrono::fsi::Real3; Pointer = thrust::device_ptr<chrono::fsi::Real3>; Derived = thrust::device_reference<chrono::fsi::Real3>; thrust::reference<Element, Pointer, Derived>::value_type = chrono::fsi::Real3]’
/opt/cuda/include/thrust/detail/reference.inl:122:38:   required from ‘thrust::reference<Element, Pointer, Derived>::operator thrust::reference<Element, Pointer, Derived>::value_type() const [with Element = chrono::fsi::Real3; Pointer = thrust::device_ptr<chrono::fsi::Real3>; Derived = thrust::device_reference<chrono::fsi::Real3>; thrust::reference<Element, Pointer, Derived>::value_type = chrono::fsi::Real3]’
/home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiInterface.cpp:99:66:   required from here
/opt/cuda/include/thrust/detail/reference.inl:137:19: error: could not convert ‘thrust::system::detail::generic::get_value<thrust::cuda_cub::tag, thrust::device_ptr<chrono::fsi::Real3> >((*(thrust::execution_policy<thrust::cuda_cub::tag>*)(& thrust::detail::derived_cast<thrust::cuda_cub::tag>((*(thrust::detail::execution_policy_base<thrust::cuda_cub::tag>*)(& non_const_system))))), ((const thrust::reference<chrono::fsi::Real3, thrust::device_ptr<chrono::fsi::Real3>, thrust::device_reference<chrono::fsi::Real3> >*)this)->thrust::reference<chrono::fsi::Real3, thrust::device_ptr<chrono::fsi::Real3>, thrust::device_reference<chrono::fsi::Real3> >::m_ptr)’ from ‘void’ to ‘thrust::reference<chrono::fsi::Real3, thrust::device_ptr<chrono::fsi::Real3>, thrust::device_reference<chrono::fsi::Real3> >::value_type’ {aka ‘chrono::fsi::Real3’}
   return get_value(thrust::detail::derived_cast(non_const_system), m_ptr);
          ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/cuda/include/thrust/system/detail/generic/memory.inl:22,
                 from /opt/cuda/include/thrust/system/detail/generic/memory.h:71,
                 from /opt/cuda/include/thrust/detail/reference.inl:22,
                 from /opt/cuda/include/thrust/detail/reference.h:173,
                 from /opt/cuda/include/thrust/memory.h:25,
                 from /opt/cuda/include/thrust/device_ptr.h:25,
                 from /opt/cuda/include/thrust/device_malloc_allocator.h:25,
                 from /opt/cuda/include/thrust/device_vector.h:25,
                 from /home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiDataManager.cuh:24,
                 from /home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiInterface.h:22,
                 from /home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiInterface.cpp:18:
/opt/cuda/include/thrust/system/detail/generic/memory.inl: In instantiation of ‘void thrust::system::detail::generic::get_value(thrust::execution_policy<Derived>&, Pointer) [with DerivedPolicy = thrust::cuda_cub::tag; Pointer = thrust::device_ptr<chrono::fsi::Real3>]’:
/opt/cuda/include/thrust/detail/reference.inl:137:19:   required from ‘thrust::reference<Element, Pointer, Derived>::value_type thrust::reference<Element, Pointer, Derived>::strip_const_get_value(const System&) const [with System = thrust::cuda_cub::tag; Element = chrono::fsi::Real3; Pointer = thrust::device_ptr<chrono::fsi::Real3>; Derived = thrust::device_reference<chrono::fsi::Real3>; thrust::reference<Element, Pointer, Derived>::value_type = chrono::fsi::Real3]’
/opt/cuda/include/thrust/detail/reference.inl:105:54:   required from ‘thrust::reference<Element, Pointer, Derived>::value_type thrust::reference<Element, Pointer, Derived>::convert_to_value_type(System*) const [with System = thrust::cuda_cub::tag; Element = chrono::fsi::Real3; Pointer = thrust::device_ptr<chrono::fsi::Real3>; Derived = thrust::device_reference<chrono::fsi::Real3>; thrust::reference<Element, Pointer, Derived>::value_type = chrono::fsi::Real3]’
/opt/cuda/include/thrust/detail/reference.inl:122:38:   required from ‘thrust::reference<Element, Pointer, Derived>::operator thrust::reference<Element, Pointer, Derived>::value_type() const [with Element = chrono::fsi::Real3; Pointer = thrust::device_ptr<chrono::fsi::Real3>; Derived = thrust::device_reference<chrono::fsi::Real3>; thrust::reference<Element, Pointer, Derived>::value_type = chrono::fsi::Real3]’
/home/bartus/AUR/chronoengine/src/chronoengine/src/chrono_fsi/ChFsiInterface.cpp:99:66:   required from here
/opt/cuda/include/thrust/detail/static_assert.h:71:7: error: invalid application of ‘sizeof’ to incomplete type ‘thrust::detail::STATIC_ASSERTION_FAILURE<false>’
       sizeof(::thrust::detail::STATIC_ASSERTION_FAILURE< (bool)( B ) >)>\
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/cuda/include/thrust/detail/static_assert.h:71:7: note: in definition of macro ‘THRUST_STATIC_ASSERT’
       sizeof(::thrust::detail::STATIC_ASSERTION_FAILURE< (bool)( B ) >)>\
       ^~~~~~
[ 96%] Building CXX object src/demos/opengl/CMakeFiles/demo_GL_cohesion.dir/demo_GL_cohesion.cpp.o
[ 96%] Linking CXX executable ../../../bin/demo_GL_shapes
```
previous gcc (7.3) works fine.

bartoszek, 2018-06-13:
Moreover gcc complains about use of GLM without defining GLM_ENABLE_EXPERIMENTAL macro.
[glm.patch.txt](https://github.com/projectchrono/chrono/files/2098480/glm.patch.txt)

saullocastro, 2018-08-02:
I get the same error using gcc-7 and cuda 9.2 on Ubuntu 16.04

thepianoboy, 2018-08-15:
commit c44d7da resolves this issue on both Arch Linux (2018-08-14) and Ubuntu 16.04

closed 2018-08-15 by thepianoboy

saullocastro, 2018-08-16:
@thepianoboy  thank you! It could compile it now :)