projectchrono/chrono #129: Improve exception safety with smart pointers

elfring, 2018-08-15:
Would you like to [wrap any pointer data members](https://github.com/projectchrono/chrono/blob/930606bde27a1c25bec371f7b73fb88f112d47c6/src/chrono/solver/ChVariablesGeneric.cpp#L22 "Update candidate: ChVariablesGeneric class") with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

rserban, 2018-08-18:
I assume you are talking about private data members in which case I am not so sure it is worth using unique_ptr.  For data members that are made public (e.g., through accessors), it's almost always the case that a shared_ptr is needed (unless you found a case where that's not true?).  Having said that, there are still places in Chrono where raw pointers are exposed to the user -- we fix these as we go along whenever we find the time.

elfring, 2018-08-18:
I suggest to [achieve safe resource management](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "C++ core guideline “R.11: Avoid calling new and delete explicitly”").