projectchrono/chrono #136: Feature Request: Vehicle in real time(soft)

arpio, 2018-10-21:
I'm still working on my c++ skills and studying the chrono docs.

How close to real time can chrono simulate a vehicle? All the demos are in "slow motion".
As far as I understand, chrono is designed more for an accurate simulation and less for real time(games)?

rserban, 2018-10-21:
This is a difficult (if not impossible) question to answer.  
You are correct that Chrono is not designed as a real time game engine.  Having said that, a simulation of one (maybe two) vehicles can run in (or close to) real time on a commodity machine.   Of course, it all depends on many factors, including your hardware, the complexity of the vehicle and scene models (physics and graphics), the solver settings, etc, etc.
How slow is slow in your case for something like demo_VEH_HMMWV9?

arpio, 2018-10-21:
I don't think the simulation is slow, the CPU step is around 30ms.

Like I said before I'm still a beginner in programming. I thought I just increase the simulation stepsize but then the vehicle brakes apart.

yumianhuli2, 2020-01-07:
> This is a difficult (if not impossible) question to answer.
> You are correct that Chrono is not designed as a real time game engine. Having said that, a simulation of one (maybe two) vehicles can run [...]

Why not live like the phyx vehicle run in realtime?

rserban, 2020-01-07:
@yumianhuli2, not sure what answer you expect here...

closed 2020-01-07 by rserban