projectchrono/chrono #140: Can't get pyChono demos to run

jlittle576, 2019-02-07:
I am on Windows 10, Anaconda 3 v 2018.12 (latest)

I tried to install via **_conda install -c projectchrono pychrono_** via Anaconda Console (ran as admin) and got

**(base) C:\Windows\system32>conda install -c projectchrono pychrono
Collecting package metadata: done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current cha
nnels:

  - pychrono**

I then tried **_conda install -c projectchrono/label/develop pychrono_** and it installed without error but upon running python line **_import pychrono as chrono_** give the following error

**_C:\ProgramData\Anaconda3\python.exe C:/wd/cd/jutil/chrono_demos/demo_masonry.py
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\core.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\ProgramData\Anaconda3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: DLL load failed: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/wd/cd/jutil/chrono_demos/demo_masonry.py", line 18, in <module>
    import pychrono.core as chrono
  File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\__init__.py", line 7, in <module>
    from .core import *
  File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\core.py", line 17, in <module>
    _core = swig_import_helper()
  File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\core.py", line 16, in swig_import_helper
    return importlib.import_module('_core')
  File "C:\ProgramData\Anaconda3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_core'_**

jlittle576, 2019-02-07:
If I open a cmd window and navigate to my Anaconda 3 scripts directory and run `conda install -c projectchrono pychrono` I get
'''
C:\wd\py3\Scripts>conda install -c projectchrono pychrono
Collecting package metadata: failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://repo.anaconda.com/pk
gs/free/win-64/repodata.json.bz2>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.

If your current network has https://www.anaconda.com blocked, please file
a support request with your network engineering team.

SSLError(MaxRetryError('HTTPSConnectionPool(host=\'repo.anaconda.com\', port=443
): Max retries exceeded with url: /pkgs/free/win-64/repodata.json.bz2 (Caused by
 SSLError("Can\'t connect to HTTPS URL because the SSL module is not available."
))'))
'''

tasora, 2019-02-07:
Dear jlittle576

Foreword: the pychrono anaconda installer is in beta stage, so there may be some issues, as we just released it yesterday.

Note: the default "conda install -c projectchrono pychrono" for the moment installs a "main" branch that is lagging behind some weeks. The other command you used (for the "develop" branch label) is what I'd suggest, as it is the most 'up to date'.

This said, I do not know why you had that issue, looks like the required  core.pyd library  (and/or one of the .dll on which it depends, ex. ChronoEngine.dll) is not found on your paths.
Can you find the ...\Anaconda3\Lib\site-packages\pychrono folder and the _core.pyd , _postprocess.pyd etc. files inside it?
Can you find the  ...\Anaconda3\Library\bin folder  and the ChronoEngine.dll  ChronoEngine_postprocess.dll etc. files inside it?

We'lll try to reproduce the error.

thanks for reporting

Alessandro Tasora

Il 07/02/2019 18:36, jlittle576 ha scritto:

I am on Windows 10, Anaconda 3 v 2018.12 (latest)

I tried to install via conda install -c projectchrono pychrono via Anaconda Console (ran as admin) and got

**(base) C:\Windows\system32>conda install -c projectchrono pychrono
Collecting package metadata: done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current cha
nnels:

  *   pychrono**

I then tried conda install -c projectchrono/label/develop pychrono and it installed without error but upon running python line import pychrono as chrono give the following error

**C:\ProgramData\Anaconda3\python.exe C:/wd/cd/jutil/chrono_demos/demo_masonry.py
Traceback (most recent call last):
File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\core.py", line 14, in swig_import_helper
return importlib.import_module(mname)
File "C:\ProgramData\Anaconda3\lib\importlib_init.py", line 127, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
File "", line 1006, in _gcd_import
File "", line 983, in _find_and_load
File "", line 967, in _find_and_load_unlocked
File "", line 670, in _load_unlocked
File "", line 583, in module_from_spec
File "", line 1043, in create_module
File "", line 219, in _call_with_frames_removed
ImportError: DLL load failed: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "C:/wd/cd/jutil/chrono_demos/demo_masonry.py", line 18, in
import pychrono.core as chrono
File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono_init_.py", line 7, in
from .core import *
File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\core.py", line 17, in
_core = swig_import_helper()
File "C:\ProgramData\Anaconda3\lib\site-packages\pychrono\core.py", line 16, in swig_import_helper
return importlib.import_module('core')
File "C:\ProgramData\Anaconda3\lib\importlib_init.py", line 127, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'core'**

Benatti1991, 2019-02-07:
Dear jlittle576,
I discourage installing packages as administrator with any virtual environment platoform (such as Anaconda or Virtualenv). Create an environment **without** running as administrator and install the proper packages (again, without running as administrator). It could cause some troubles mixing the system Python with the packaged python.
Type `conda install -c projectchrono/label/develop pychrono` if you are using Windows.
Let us know if you have any problem.
Simone

jlittle576, 2019-02-07:
> Dear jlittle576 Foreword: the pychrono anaconda installer is in beta stage, so there may be some issues, as we just released it yesterday. 

Note: the default "conda install -c projectchrono pychrono" for the moment installs a "main" branch that is lagging behind some weeks. The other command you used (for the "develop" branch label) is what I'd suggest, as it is the most 'up to date'. 

This said, I do not know why you had that issue, looks like the required core.pyd library (and/or one of the .dll on which it depends, ex. ChronoEngine.dll) is not found on your paths. 

Can you find the ...\Anaconda3\Lib\site-packages\pychrono folder and the _core.pyd , _postprocess.pyd etc. files inside it? 

Can you find the ...\Anaconda3\Library\bin folder and the ChronoEngine.dll ChronoEngine_postprocess.dll etc. files inside it? We'lll try to reproduce the error. thanks for reporting Alessandro Tasora Il >.

Thanks for the reply. I am using the develop branch, it is the only one that will install. All of those files are present in my Anaconda3 directory. If you need any additional debugger/state information let me know.

> I discourage installing packages as administrator with any virtual environment platform (such as Anaconda or Virtualenv). Create an environment **without** running as administrator and install the proper packages (again, without running as administrator). It could cause some troubles mixing the system Python with the packaged python>.

I am using a standard windows install of Anaconda3, not a virtual environment. I run the conda install command in the Anaconda Console that is specific to that install. If I don't run that console as admin, the install fails due to lack of permissions.

Benatti1991, 2019-02-07:
installing Anaconda "for all users" will put it in a directory in which you need admin privileges to write. Try installing Anaconda just for you, it will be put in a usr path avoiding a lot of issues.
BTW, Anaconda creates virtual environments, are you installing in one of these or in the base env?

jlittle576, 2019-02-07:
> installing Anaconda "for all users" will put it in a directory in which you need admin privileges to write. Try installing Anaconda just for you, it will be put in a usr path avoiding a lot of issues.
> BTW, Anaconda creates virtual environments, are you installing in one of these or in the base env?

I guess it would be base env. We are the end users of our code so I don't have a need for virtual env as far as I know. I do "for all users" so Anaconda installs to C:/ProgramData, this is how we have been installing Anaconda for awhile and its worked fine so far. We only have 1 user per machine. Windows doesn't have a default python install (at least not one that's in the path) so I'm not sure what it would conflict with.

I'll try a user install as a debug step to see if that resolves the issue.

Benatti1991, 2019-02-08:
Installing packages in an environment different from _base_ is generally advisable. Let us know how it went.

jlittle576, 2019-02-11:
Hey guys. I tried the following steps

- New Anaconda3 install with all default option (User install)
- Cloned base into a new env called tstenv
- Ran `conda install --name tstenv -c projectchrono/label/develop pychrono`
- Opened Spyder (using the shortcut created for tstenv)
- Tried to run the demo files

Here are the results:

- demo_python_1.py runs without error
- demo_fea_beams.py gives `ModuleNotFoundError: No module named 'pychrono.mkl'`
- most other demos with Irrlicht visualization just give an all black window (but no exceptions)

Benatti1991, 2019-02-11:
Regarding the mkl module is perfectly normal, it hasn't been added yet.
I can't figure out the odd thing you're experiencing with irrlicht. Could you please try to launch from the anaconda prompt? Sometimes irrlicht error are not displayed in spyder prompt.
Btw, why did you clone base environment instead of creating a new one?
Last thing, which python version are you using?

jlittle576, 2019-02-11:
Yeah, 3.7.1, from Anaconda 2018.12 (latest)

I don't know much about envs since I always just use Anaconda base install. Figured I needed to clone base for all the libs but now realize it just auto-installs whatever is needed, cool cool cool.

I created a new env, installed pychrono, and ran demo_masonry.py from the console along with the black window I got:

```
Could not load shader function D3DXAssembleShader from dll, shaders disabled: d3
dx9_43.dll
Could not create vertex shader.
Could not create vertex shader.
Resizing window (1024 768)
Resetting D3D9 device.
Could not load shader function D3DXCompileShader from dll, shaders disabled: d3d
x9_43.dll
Could not load font because the file does not exist: ../../../data/fonts/arial8.
xml
Could not open file of texture: ../../../data/skybox/sky_lf.jpg
Could not open file of texture: ../../../data/skybox/sky_dn.jpg
Could not open file of texture: ../../../data/skybox/sky_up.jpg
Could not open file of texture: ../../../data/logo_pychrono_alpha.png
Could not open file of texture: ../../../data/concrete.jpg
Could not open file of texture: ../../../data/concrete.jpg
Could not open file of texture: XEFFECTS_SM_512
XEffects: Please ignore previous warning, it is harmless.
Could not open file of texture: XEFFECTS_SM_512_2
XEffects: Please ignore previous warning, it is harmless.
```

Benatti1991, 2019-02-11:
Ok, thank you! Now I can't look further into it (I'm home sick) so could you please try creating a new environment with python 3.6? Maybe we could narrow down the list.

jlittle576, 2019-02-11:
Yeah no problem. Looks like the same result for 3.6.

```
Could not load shader function D3DXAssembleShader from dll, shaders disabled: d3
dx9_43.dll
Could not create vertex shader.
Could not create vertex shader.
Resizing window (1024 768)
Resetting D3D9 device.
Could not load shader function D3DXCompileShader from dll, shaders disabled: d3d
x9_43.dll
Could not load font because the file does not exist: ../../../data/fonts/arial8.
xml
Could not open file of texture: ../../../data/skybox/sky_lf.jpg
Could not open file of texture: ../../../data/skybox/sky_dn.jpg
Could not open file of texture: ../../../data/skybox/sky_up.jpg
Could not open file of texture: ../../../data/logo_pychrono_alpha.png
Could not open file of texture: ../../../data/concrete.jpg
Could not open file of texture: ../../../data/concrete.jpg
Could not open file of texture: XEFFECTS_SM_512
XEffects: Please ignore previous warning, it is harmless.
Could not open file of texture: XEFFECTS_SM_512_2
XEffects: Please ignore previous warning, it is harmless.
```

demo_python_2.py does run so at least it looks like the core simulation engine is running, pretty cool stuff.

dariomangoni, 2019-02-12:
I'm not into PyChrono, but I would suggest to give a look to the [ProjectChrono](https://projectchrono.org/forum/) forum.
Especially this post that I wrote some time ago. It seems quite close to your problem:
[https://projectchrono.org/forum/?place=msg%2Fprojectchrono%2FHKOgv-mdzz4%2FSrcSo_H0CgAJ](https://projectchrono.org/forum/?place=msg%2Fprojectchrono%2FHKOgv-mdzz4%2FSrcSo_H0CgAJ)

As it is written there, try to disable shadows for now.

jlittle576, 2019-02-12:
Thanks! Luckily, since I'm working from python, solution 1 (installed DirectX libraries) works for me.

tasora, 2019-02-13:
Hi,

I may add a comment here: many of those errors from the Irrlicht 
visualization system are likely to be caused by the fact that you are 
running the .py file from a directory for whom the data/ directory (with 
fonts, textures, assets, meshes, etc.) is not at relative path 
../../../data.

By default, Chrono assumes the data directory is at |../../../data/|

You can change that by using, for instance,

    chrono.SetChronoDataPath("../data/")

or such. I do not remember where the data folder is (respect to the py 
demos) when you use the Anaconda installer, because I use the the C++ 
full repository that has data at ../../../data. But you can easily 
figure it and fix this kind of problems by adjusting the 
chrono.SetChronoDataPath(......)  at the beginning of your python program

Also note that if you run the py files from the VisualStudio Python, the 
current directory is not the directory of the py file. In Spyder, it is.

Other than this, the black screen issue depends on video drivers. Try 
installing recent DirectX libs or disable lights with shadows, as Dario 
suggested. For example, on OSX this was always an issue.

Alessandro

jlittle576, 2019-02-13:
Simply installing the DirectX libraries in Dario's link got the visualization working.

Thanks for the tip on the textures. I figured the "Please ignore warning" applied to theses messages as well. I set the path and now it works with textures. Thanks for the help!

I also figured out that the reason it wouldn't run in pyCharm was a path issue. I didn't let Anaconda3 change the PATH variable when I installed to user.

Finally, I had to change 'LoadWavefrontMesh(chrono.GetChronoDataPath() +'body_1_1.obj')' to 'LoadWavefrontMesh(chrono.GetChronoDataPath() +'solid_works/swiss_escapement_shapes/body_1_1.obj')' to get the mesh_demo to run.

closed 2019-02-26 by dariomangoni

Benatti1991, 2019-02-27:
@jlittle576  since maybe you need the fea module I just added the MKL module to the conda package, and it will be ready for download in a few hours.

BeinerF, 2020-08-12:
I have basically the same problem described above by @jlittle576 . 

**my system:**
`OS Name	Microsoft Windows 10 Pro
Version	10.0.18362 Build 18362
System Manufacturer	LENOVO
System Type	x64-based PC
Processor	Intel(R) Core(TM) i7-3610QM CPU @ 2.30GHz, 2301 Mhz, 4 Core(s), 8 Logical Processor(s)
BIOS Version/Date	LENOVO G5ETB6WW (2.76 ), 9/09/2019
Hardware Abstraction Layer	Version = "10.0.18362.752"
Installed Physical Memory (RAM)	20.0 GB
Kernel DMA Protection	Off
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualisation Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes
`

**I tried multiple times:**
- clean user anaconda 3 install
- created a new clean python 3.7 environment
- run `conda install -c projectchrono/label/develop pychrono`
- tried one of the basic examples: `python demo_CH_EulerAngles.py`

**result as mentioned above:**
`
(ChronoEnv) C:\Users\frank\anaconda3\envs\ChronoEnv\Lib\site-packages\pychrono\demos\core>python demo_CH_EulerAngles.py
Traceback (most recent call last):
  File "C:\Users\frank\anaconda3\envs\ChronoEnv\lib\site-packages\pychrono\core.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "C:\Users\frank\anaconda3\envs\ChronoEnv\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: DLL load failed: A dynamic link library (DLL) initialization routine failed.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "demo_CH_EulerAngles.py", line 11, in <module>
    import pychrono as chrono
  File "C:\Users\frank\anaconda3\envs\ChronoEnv\lib\site-packages\pychrono\__init__.py", line 37, in <module>
    from .core import *
  File "C:\Users\frank\anaconda3\envs\ChronoEnv\lib\site-packages\pychrono\core.py", line 17, in <module>
    _core = swig_import_helper()
  File "C:\Users\frank\anaconda3\envs\ChronoEnv\lib\site-packages\pychrono\core.py", line 16, in swig_import_helper
    return importlib.import_module('_core')
  File "C:\Users\frank\anaconda3\envs\ChronoEnv\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_core'

(ChronoEnv) C:\Users\frank\anaconda3\envs\ChronoEnv\Lib\site-packages\pychrono\demos\core>`

I tried various python version, multiple anaconda reinstalls, etc..
I am really keen to give pychrono a go, but I seem to be out of luck. :(

BeinerF, 2020-08-27:
I switched to building projectchrono(incl. pychrono) from source.
Solved my above described problems.
Thx for this amazing project.