projectchrono/chrono #145: Install on Windows

NingningSong, 2019-03-17:
Hi, recently, I try to install the projectchrono on my computer, but when I build the parallel module, I find that my computer needs to install the boost.lib and the blaze.lib, I follow the instructions of the website and download the boost.lib and blaze.lib, but I do not clear how to install and configure the boost.lib and blaze.lib. Therefore, I want to consult you how could I install and configure the boost.lib and the blaze.lib to compile and pass.
      I am appreciated if you can help me install the projectchrono.    
best wishes to you

rserban, 2019-03-17:
You do not need to build any of those two libraries.  Blaze is headers-only and Boost is only needed for Blaze.  All you need to do is download the appropriate archives, extract them on your disk in locations of your choosing, and then point to those locations during the Chrono CMake configuration.

For Boost, you should be able to use the latest stable release (1.69) from https://www.boost.org/users/download/

For Blaze, **do not get the latest release** as it will not work with Chrono on Windows.  Instead, get version 3.0 (from https://bitbucket.org/blaze-lib/blaze/downloads/).

Note that the Chrono::Parallel module also requires the Thrust library.  The easiest way to get this is to install CUDA (which includes Thrust).  Otherwise, you can get the stand-alone Thrust library (from http://thrust.github.io/) but that may be a bit more complicated. Thrust is also a headers-only library, so no need to "build" it.

NingningSong, 2019-03-18:
I am very glad to receive your reply so soon, I have followed your advice to build the Chrono::Parallel module, but when I build the Chrono::Parallel on the page of Cmake-gui, I find the following questions:

[image]

And then I ignore this hint and press generate until all the variables get a white background. But when I  build the .sln file in the visual studio 2013, then the following problems appear:

[image]

I do not clear how to deal with these problems.  So I hope I could get you help.
best wishes

rserban, 2019-03-18:
The problem is that Chrono requires a newer version of Visual Studio, with more complete C++11 support.  VS2015 should work, but you might as well jump to VS2017. The community edition is available for free (https://visualstudio.microsoft.com/downloads/).

I now see that we failed to update our installation page to reflect this. We will fix that.

As for MPI, that is indeed not a requirement for Chrono::Parallel which uses OpenMP.  You will need MPI if you want to try out the Chrono::Distributed module.  For Windows, you can get the Microsoft library MS-MPI (https://www.microsoft.com/en-us/download/details.aspx?id=57467).

NingningSong, 2019-03-18:
Thank you very much to receive you advice, I will follow your advice to rebuild the Chrono::Parallel.
Best wishes to you!

NingningSong, 2019-03-18:
Dear professor rserban, I have followed your advice to install visual studio 2015 on my disk, but when I build in the Cmake-gui, then the Cmake-gui appear  could not found any instance of visual. How should I do to solve this problems:
[image]
best wishes

NingningSong, 2019-03-18:
Sorry to disturb you so many, I have solved this problem on visual studio 2017.
Best wishes to you

rserban, 2019-03-18:
No problem.  Happy to hear you got it to work.

closed 2019-03-18 by rserban