projectchrono/chrono #147: Benchmark Testing

acxz, 2019-03-31:
There are many physics engines out there for the simulation/robotics community. As such, there is a need for benchmarking various engines for accurate and objective comparisons. Benchmarking also helps end users determine which is the appropriate engine for their task and for developers to help debug and better their own engines.

There is ongoing work for benchmarking at [SimBenchmark](https://github.com/leggedrobotics/SimBenchmark). As of now RaiSim, Bullet, ODE, MuJoCo, and DART have been benchmarked ([results](https://leggedrobotics.github.io/SimBenchmark/)), but it would be great to have more engines benchmarked.

As such, I would like to invite members of the CHRONO community to help test & benchmark CHRONO under the same framework. Based on my (limited) knowledge, this is the first time such benchmarking has been openly available. Utilizing this instead of performing closed benchmarking is beneficial for both the robotics community as well as the CHRONO community.

tasora, 2019-04-03:
hi

thank you for pointing out this opportunity. At the moment we are busy with other developments, so we have to postpone benchmarking. In detail, we plan to spend next months in optimizing the speed of Chrono, so benchmarking must be a later priority.

thanks

Alessandro

Il 3/31/2019 7:06 PM, acxz ha scritto:

There are many physics engines out there for the simulation/robotics community. As such, there is a need for benchmarking various engines for accurate and objective comparisons. Benchmarking also helps end users determine which is the appropriate engine for their task and for developers to help debug and better their own engines.

There is ongoing work for benchmarking at SimBenchmark<https://github.com/leggedrobotics/SimBenchmark>. As of now RaiSim, Bullet, ODE, MuJoCo, and DART have been benchmarked (results<https://leggedrobotics.github.io/SimBenchmark/>), but it would be great to have more engines benchmarked.

As such, I would like to invite members of the CHRONO community to help test & benchmark CHRONO under the same framework. Based on my (limited) knowledge, this is the first time such benchmarking has been openly available. Utilizing this instead of performing closed benchmarking is beneficial for both the robotics community as well as the CHRONO community.

acxz, 2019-04-03:
I understand. I wish you all the best. I can't wait to see the results!