projectchrono/chrono #150: PyChrono for Mac

laflij, 2019-04-08:
A pychrono package for Mac would be great!

Benatti1991, 2019-04-23:
We build anaconda packages using Appveyor, which is not supporting Mac VM so far. 
You can always download and build the source code, it's a bit more complicated, but it's well documented on our guide (http://api.projectchrono.org/development/tutorial_table_of_content_install.html).
Simone

closed 2019-06-22 by Benatti1991

pengzhenghao, 2020-08-31:
For your information, now we can install pychrono in Mac via:

Installing chrono from the conda-forge channel can be achieved by adding conda-forge to your channels with:

```
conda config --add channels conda-forge
conda install chrono pychrono
```

I found this in: https://github.com/conda-forge/pychrono-feedstock

Benatti1991, 2020-08-31:
Peng,
actually PyChrono for Mac is now available from our Anaconda channel, just type:
`conda install -c projectchrono/label/develop pychrono`
conda-forge package misses some important features, such as Irrlicht rendering and MKL solver.
Simone

laflij, 2020-08-31:
Great news, thanks for letting me know!

nonlosoenonlovogliosapere, 2020-10-29:
hi, i m trying to install pychrono on Osx from the channel projectchrono/label/develop using a conda environment, but after the installation i get the following error on a python console (the conda env is correctly activated)
```
import pychrono as chrono
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<CONDA_ENV_PATH>/lib/python3.8/site-packages/pychrono/__init__.py", line 40, in <module>
    from .core import *
  File "<CONDA_ENV_PATH>/lib/python3.8/site-packages/pychrono/core.py", line 13, in <module>
    from . import _core
ImportError: dlopen(<CONDA_ENV_PATH>/lib/python3.8/site-packages/pychrono/_core.so, 2): Library not loaded: /usr/local/opt/libomp/lib/libomp.dylib
  Referenced from: <CONDA_ENV_PATH>/lib/python3.8/site-packages/pychrono/_core.so
  Reason: image not found
```

Benatti1991, 2020-10-29:
Hi,
please install libomp through brew:
`brew install libomp `
Ideally the package should use libomp from Anaconda, I'm working on it but I still haven't found a way to get it done. In the meantime that fix should work.
Simone.

danimspaulino, 2022-03-08:
Dear Simone,

I am installing pychrono in MacOS and I am getting the following error:
ModuleNotFoundError: No module named 'pychrono.pardisomkl'

Same happens for the visualization package Irrlicht. 

Do you have any suggestions on what could it be? I don't know if matters, but I'm using M1 chip macbook air. 

Best,

Dani