projectchrono/chrono #165: Illegal hardware instruction while importing pychrono (linux)

zambardus, 2019-06-07:
Hi, I have a mac o/s machine but since there is no direct installation support for it, I borrowed a machine with linux mint o/s and I was able to install pychrono v4.0.0. But while running python file, I get this error "Illegal hardware instruction pychrono" corresponding to the line `import pychrono as chrono` I have seen similar issue with tensor flow where people had to migrate to a lower version. Can anyone help me with this? Or let me know if I need to move to a lower version of pychrono.

Benatti1991, 2019-06-10:
It seems to be related to AVX instructions missing in your CPU. Try typing in your shell the following line:
`$ more /proc/cpuinfo | grep flags`
And please report the output (see https://tech.amikelive.com/node-887/how-to-resolve-error-illegal-instruction-core-dumped-when-running-import-tensorflow-in-a-python-program/).
If the problem is indeed what I supposed you could always build Chrono from source (even on you mac PC) with Python module enabled. Here you find instruction:

http://api.projectchrono.org/development/tutorial_install_chrono.html

http://api.projectchrono.org/development/module_python_installation.html

jaberkow, 2019-06-13:
I received a similar error on  Ubuntu 14.04.6 LTS after installing with the conda installer:
```
>>> import pychrono as chrono
Illegal instruction (core dumped)
```
Here is the more output.

```
$ more /proc/cpuinfo | grep flags
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm epb ssbd ibrs ibpb stibp kaiser tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts flush_l1d
```
(the above is repeated four times for my quad core).  Note that my cpu includes support for AVX instructions.  Any advice on how to solve this, or should I just install from source?

Benatti1991, 2019-06-13:
@jaberkow actually PyChrono uses AVX2, which is missing in your CPU flags.
As for the OpenAI Gym environments compiling form source shouldn't be a problem:  in your gym envs you want to put only the .py files with the training environments. Thy will import PyChrono regardless of how you installed the module. I also think that Roboschool (and also MuJoCo) work in a similar way.

jaberkow, 2019-06-13:
Have you been able to install pychrono on any EC2 instances on AWS?

Benatti1991, 2019-06-15:
Actually I do not have AWS credits so I've never tried. Since this is not a proper issue (and not the issue in the title) please consider opening a separate thread (maybe on the [forum ](https://projectchrono.org/forum/))

closed 2019-11-04 by Benatti1991