projectchrono/chrono #167: Question about Chrono-tensorflow

jaberkow, 2019-06-10:
It appears that the environment interface in the chrono-tensorflow demo closely emulates the OpenAI gym framework.  Is there a reason not to just subclass an OpenAI gym and use their utilities (Spaces, etc)?

Benatti1991, 2019-06-10:
Actually it is something we are planning to do, but I haven't looked further into it. 
You are right, it would be nice to have PyChrono environments inside Gym.
Since the vehicle module is now wrapped in PyChrono I could spend some time on this implementation now.
If you are willing to contribute please contact me (you should find my email adress in my GitHub profile).

Benatti1991, 2019-09-23:
UPDATE: gym-chrono has been released. We also added 2 more complex tasks (a robotic arm and a 
hexapod walker). 
The environments are fully compatible with OpenAI Gym environments and tasted with OpenAI's baselines.
https://github.com/projectchrono/gym-chrono

closed 2019-11-04 by Benatti1991