projectchrono/chrono #168: ChLinkLockCylindrical

bchambers28, 2019-06-14:
I'm setting up a simulation using the pychrono library which has two linked bodies. I would like to use a ChLinkLockCylindrical joint with the axis of rotation to be the y-axis. Is there any way to change the default axis of rotation from the z-axis to the y-axis or is it always going to be the z-axis?

Benatti1991, 2019-06-15:
ChLinkLock links (Cylindrical, Prismtic, Revolute) are always along the z axis, but this is not a limitation, since you can always rotate the coordinate system **of the link** (ChCoordsysD) in any direction, or directly initialize the Coordinate System with a quaternion with the right oreintation (look at Leg_chordsys [here ](https://github.com/projectchrono/chrono/blob/develop/src/demos/python/chrono-tensorflow/envs/chtrain_ant.py)).
There's a bunch of ready to use quaternions for axis swapping:
Q_ROTATE_Y_TO_X 
Q_ROTATE_Y_TO_Z 
Q_ROTATE_X_TO_Y 
Q_ROTATE_X_TO_Z 
Q_ROTATE_Z_TO_Y 
Q_ROTATE_Z_TO_X 
Q_FLIP_AROUND_X 
Q_FLIP_AROUND_Y 
Q_FLIP_AROUND_Z 

So in your case, yo should:
- Create the link
- Create the Coorsdys in the wanted position with the right rotation ( chrono.Q_ROTATE_Z_TO_Y I suppose) 
- initialize the link between body A, body B with the CoordSys you just created
I would suggest you to take a look to the Python demos in the github (or your local Anaconda package) folder.
Simone

bchambers28, 2019-06-17:
Thanks Simone.

closed 2019-06-17 by bchambers28