projectchrono/chrono #169: Chrono bodies moving with no force.

bchambers28, 2019-06-20:
I'm setting up a chrono system using pychrono that has gravity in the y-direction. It consist of multiple bodies in a line which are connected into a system using a ChLinkLockCylindrical link in the middle as well as a ChLinkLockRevolute link between each body. The first body is fixed and the rest are free. The only force in the system is from gravity. With only the ChLinkLockCylindrical link active the system falls under gravity and creates a pendulum like motion which is to be expected. When I add the ChLinkLockRevolute links the system bodies (with the exception of the body which is fixed) begin moving slowly in the x-direction. I have included an example of how I am setting the link. Am I just setting the linkage incorrectly or is this some other problem? 
` o_first_point = pychrono.ChCoordsysD(pychrono.ChVectorD(2,2,2))
  o_second_point = pychrono.ChCoordsysD(pychrono.ChVectorD(2,2,3))

  # Create the joint
  o_link_lock_right_land = pychrono.ChLinkLockRevolute()
  o_link_lock_right_land.Initialize(body_1, body_2, False, o_first_point, o_second_point)

  # Add it to the system
  o_chrono_system.ChSystem.Add(o_link_lock_right_land)`

Benatti1991, 2019-06-21:
@bchambers28 could you please 
- write all the code of your simulation (it's hard to understand what you are trying to achieve from your description)
- use the "code" (see the <> icon in the answer panel) to write the code, otherwise the code is difficult to read

bchambers28, 2019-06-27:
@Benatti1991 Unfortunately my code has certain portions that I cannot share. I can attach thee portion of code specifically for the linkages if that would be helpful? I have found other instances of this occurring in the google group and have been trying some of the changes those people used or suggested currently.

Benatti1991, 2019-06-27:
Many problems pointed in the google group are related to odd mass ratios, large timesteps and stuff like that, but since your problem pops out when you add a link I think that it is maybe due to some error in constraint definition so yes, attaching that portion of code would be helpful. 
Alternatively, you can sand me the code privately via mail.

closed 2026-03-10 by rserban