projectchrono/chrono #172: Pychrono import issues

Bawaw, 2019-07-29:
Hey, 

I'm having trouble installing Pychrono on ubuntu 18.04. First I tried installing via conda but I ended up with 'Illegal instruction (core dumped)' error, this is probably related to this issue [https://github.com/projectchrono/chrono/issues/165](url). So now I'm trying to built from source. 

I've followed the instructions on the website but whenever I try to import the pychrono library I get the following error: 
```>>> import pychrono
Traceback (most recent call last):
  File "/media/alpha/Projects/chrono/build/bin/pychrono/core.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/home/bawaw/anaconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pychrono._core'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/media/alpha/Projects/chrono/build/bin/pychrono/__init__.py", line 7, in <module>
    from .core import *
  File "/media/alpha/Projects/chrono/build/bin/pychrono/core.py", line 17, in <module>
    _core = swig_import_helper()
  File "/media/alpha/Projects/chrono/build/bin/pychrono/core.py", line 16, in swig_import_helper
    return importlib.import_module('_core')
  File "/home/bawaw/anaconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: dynamic module does not define module export function (PyInit__core)
```

The same happens when I try to run one of the python demos. Any idea what is going on here? 

Kind regards, 
Balder

Benatti1991, 2019-09-22:
Sorry for the late answer, we figured out that it is a problem related to SWIG 4.0. We are fixing the issue, in the meantime you could 
- Downgrade to SWIG 3.0.12
OR
- Copy and paste the _core.so, _irrlicht.so etc in the pychrono folder (now they are up one level)

Benatti1991, 2019-09-24:
The latest update fixed the issue. Please let me know if you still have problems.

closed 2019-10-01 by Benatti1991