projectchrono/chrono #179: Pychrono tutorials do not succeed

felixvd, 2019-09-04:
The first two Pychrono tutorials 

demo_python_2.py fails with:

```
time= 0.45000000000000023  bodyB y= 0.9982750315016988
Traceback (most recent call last):

  File "<ipython-input-1-31da8eea56df>", line 1, in <module>
    runfile('/home/felix/dev/pychrono-tutorials/demo_python_2.py', wdir='/home/felix/dev/pychrono-tutorials')

  File "/home/felix/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 704, in runfile
    execfile(filename, namespace)

  File "/home/felix/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 108, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/felix/dev/pychrono-tutorials/demo_python_2.py", line 90, in <module>
    my_system.GetContactContainer().ReportAllContacts(my_rep)

  File "/home/felix/anaconda3/lib/python3.7/site-packages/pychrono/core.py", line 16530, in ReportAllContacts
    return _core.ChContactContainer_ReportAllContacts(self, *args)

RuntimeError: SWIG director method error. Error detected when calling 'ReportContactCallback.OnReportContact'
```
The "Report Contact Callback" and "Simulation loop" sections should be commented out until the "Add contact callback" is fixed.

demo_python_1.py fails on the line `ma.SetMatr(mlist)` with this error:

```
[...]
quat. product= [ 30, 0, 0, 0 ]
Traceback (most recent call last):

  File "<ipython-input-3-80a9b323cce9>", line 1, in <module>
    runfile('/home/felix/dev/pychrono-tutorials/demo_python_1.py', wdir='/home/felix/dev/pychrono-tutorials')

  File "/home/felix/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 704, in runfile
    execfile(filename, namespace)

  File "/home/felix/anaconda3/lib/python3.7/site-packages/spyder_kernels/customize/spydercustomize.py", line 108, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/home/felix/dev/pychrono-tutorials/demo_python_1.py", line 53, in <module>
    ma.SetMatr(mlist)   # Create a Matrix from a list. Size is adjusted automatically.

AttributeError: 'ChMatrixDynamicD' object has no attribute 'SetMatr'
```

If someone can give a hint about how to fix that issue, I can submit a PR along with some other minor fixes.

Benatti1991, 2019-09-10:
Dear Felix,
these demos were recently  changed according to the API changes. Are you compiling pychrono from source? If yes, please try REbuilding the solution.

felixvd, 2019-09-10:
Thanks for the message. No, I am just using Python for quick prototyping. It's no problem, I will just wait for a new release.

Benatti1991, 2019-09-10:
So are you installing from Anaconda? The latest anaconda build (_402) works for me.

felixvd, 2019-09-10:
I did install it in an anaconda environment, but I forgot how exactly or where to look up the version number. Feel free to close if you say this should be solved.

Benatti1991, 2019-09-10:
You could uninstall and reinstall the package. When you reinstall and anaconda asks for confirmation (y/n) the version number is shown
Simone

Benatti1991, 2019-09-13:
Which OS are you using?

ZhiquanW, 2019-09-13:
* The second issue : AttributeError: 'ChMatrixDynamicD' object has no attribute 'SetMatr'
* My OS is Ubuntu 18.04.3lts

Benatti1991, 2019-09-13:
I just tried and it throws no error. It seems that your demos feature methods that I recently added while your PyChrono doesn't. Have you built pychrono from sources and set the PYTHONPATH variable in the past? Have you installed PyChrono in your base environment? I'm thinking about possible explanations.

ZhiquanW, 2019-09-13:
Thank you for your reply! I install PyChrono from anaconda, no from sources.

Benatti1991, 2019-09-13:
Have you installed it both in your base environment and new environment?

ZhiquanW, 2019-09-14:
Thank you for your help!!!! I run it successfully in the base environment !!!

Benatti1991, 2019-09-16:
I generally recommend to avoid installing packages in the base environment, to avoid issues like this (and worse). 
In your shoes I would uninstall pychrono in the base env and do everything in a new environment.

ZhiquanW, 2019-09-16:
That is a good idea. I will follow your suggestion, as i'm new to anaconda and know little about that.

Benatti1991, 2019-09-16:
@felixvd is the issue solved for you too? Can I close?

felixvd, 2019-09-17:
Thanks by the way!

closed 2019-09-17 by felixvd