projectchrono/chrono #187: Advice on how to simulate flexible rods subject to wind (or grass with wind).

andretag, 2019-09-23:
As the title says, do you have any advice on how I could simulate one (ore more, in the order of 10) flexible rods subject to gravity and a wind force (e.g. similar to a grass with wind)? 
- I am interested in capturing the interaction (contact) between multiple rods.
- I am not too interested about the physical accuracy of the airflow/bending simulation. I would value simulation speed more.

Is Chrono the right tool for this? If yes, could you kindly point me to the best tool boxes to use?

rserban, 2019-10-01:
Yes, you shuld be able to do this with Chrono.  One possibility is to use ANCF cable elements to model the grass blades (see demo_FEA_cables) and a distributed load to mimic the effect of the wind (see demo_FEA_loads).

--Radu

P.S.  If you do end up using Chrono, please post usage questions to the user mailing list ([http://projectchrono.org/forum/](http://projectchrono.org/forum/)). You are more likely to get answers to this type of questions there.

andretag, 2019-10-03:
Hi @rserban, thank you for your reply! I will, I discovered the forum only after I had posted the question.

closed 2019-10-03 by rserban