projectchrono/chrono #191: pychrono segmentation fault on MacOS X 10.14.6

tubiana, 2019-10-16:
Hi! 

I tried to install pychrono on MacOS X 10.14.6 within a clean conda environment and when I try to import pychrono, I got a 'Segmentation fault: 11"

here's my environment : 

```
# packages in environment at /Users/thibault/anaconda3/envs/assembly:
#
# Name                    Version                   Build  Channel
appnope                   0.1.0            py36hf537a9a_0  
backcall                  0.1.0                    py36_0  
bzip2                     1.0.8                h1de35cc_0  
ca-certificates           2019.8.28                     0  
certifi                   2019.9.11                py36_0  
decorator                 4.4.0                    py36_1  
icu                       58.2                 h4b95b61_1  
intel-openmp              2019.4                      233  
ipython                   7.8.0            py36h39e3cac_0  
ipython_genutils          0.2.0            py36h241746c_0  
jedi                      0.15.1                   py36_0  
libboost                  1.67.0               hebc422b_4  
libcxx                    4.0.1                hcfea43d_1  
libcxxabi                 4.0.1                hcfea43d_1  
libedit                   3.1.20181209         hb402a30_0  
libffi                    3.2.1                h475c297_4  
libiconv                  1.15                 hdd342a3_7  
mkl                       2019.4                      233  
ncurses                   6.1                  h0a44026_1  
openssl                   1.1.1d               h1de35cc_2  
parso                     0.5.1                      py_0  
pexpect                   4.7.0                    py36_0  
pickleshare               0.7.5                    py36_0  
pip                       19.2.3                   py36_0  
prompt_toolkit            2.0.10                     py_0  
ptyprocess                0.6.0                    py36_0  
pychrono                  4.0.0                  py36_402    projectchrono
pygments                  2.4.2                      py_0  
python                    3.6.9                h359304d_0  
readline                  7.0                  h1de35cc_5  
setuptools                41.4.0                   py36_0  
six                       1.12.0                   py36_0  
sqlite                    3.30.0               ha441bb4_0  
tk                        8.6.8                ha441bb4_0  
traitlets                 4.3.3                    py36_0  
wcwidth                   0.1.7            py36h8c6ec74_0  
wheel                     0.33.6                   py36_0  
xz                        5.2.4                h1de35cc_4  
zlib                      1.2.11               h1de35cc_3  

```

Is there a quick way to fix this :) ? 
Thank you very much ! 
Thibault.

Benatti1991, 2019-10-16:
We are aware of problems with MacOS, we are trying to figure out the source of the issue but currently I don't have any Mac machine to test on. Could you please debug the core.py file and tell me which class throws the error?
Thanks,
Simone
PS: I know for sure that you would have the same issue compiling from sources

tubiana, 2019-10-16:
Hi again and thank you for your quick answer. 

I'm not sure how to debug core.py but here's a traceback if it can help...

```
(assembly) c02z31l2lvdm:misc thibault$ python -Xfaulthandler main.py 
Fatal Python error: Segmentation fault

Current thread 0x0000000102ed85c0 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 922 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 658 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1023 in _handle_fromlist
  File "/Users/thibault/anaconda3/envs/assembly/lib/python3.6/site-packages/pychrono/core.py", line 13 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/Users/thibault/anaconda3/envs/assembly/lib/python3.6/site-packages/pychrono/__init__.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "main.py", line 2 in <module>
Segmentation fault: 11
```

Best,
Thibault

peterson79, 2020-03-11:
I am running into the same problem:

```
First tutorial for PyChrono: vectors, matrices etc.
Fatal Python error: Segmentation fault

Current thread 0x0000000112a04dc0 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 922 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 658 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1023 in _handle_fromlist
  File "/Users/jasonpeterson/.conda/envs/JumpBot3_6/lib/python3.6/site-packages/pychrono/core.py", line 13 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "/Users/jasonpeterson/.conda/envs/JumpBot3_6/lib/python3.6/site-packages/pychrono/__init__.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 678 in exec_module
  File "<frozen importlib._bootstrap>", line 665 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 955 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 971 in _find_and_load
  File "JumpBotPhysics.py", line 14 in <module>
zsh: segmentation fault  python -Xfaulthandler JumpBotPhysics.py
(JumpBot3_6) jasonpeterson@Jasons-MacBook-Pro JumpBot % 
```

offending line in core.py is (13):

`from . import _core`

Can someone point me in the right direction to debug this?  I have extensive experience with C/C++ and Xcode, just not the python side of things, I am trying something new :)

Thanks! --Jason

Benatti1991, 2020-03-11:
Are you compiling from sources? Look into this issue for for details about Clang verisons know to work with SWIG and Chrono
https://github.com/projectchrono/chrono/issues/194

Simone

peterson79, 2020-03-11:
No I am not compiling from sources, I am using:

`conda install -c projectchrono pychrono`

--Jason

Benatti1991, 2020-03-11:
MacOS support for conda packages is not there yet, for the moment you should build from sources

Simone

cyndiac, 2020-07-13:
Was anyone able to build from source successfully? I still get segmentation fault:11 when I try to import pychrono. 

Cyndia

Benatti1991, 2020-07-13:
Hi Cyndia,
from what I could get from the PyChrono MacOS users using SWIG 4 and Xcode >= 11 seems to solve the issue when compiling from sources. Any feedback is appreciated.
Simone

cyndiac, 2020-07-13:
I have SWIG 4.0.2 and Xcode 11.5. I'm not sure how to get any more details about the error.

Benatti1991, 2020-07-13:
Clang version?

cyndiac, 2020-07-13:
Apple clang version 11.0.3 (clang-1103.0.32.62)

Target: x86_64-apple-darwin19.5.0

Thread model: posix

rainergericke, 2020-07-14:
The c++ part looks good. We need some information about your python installation:

please type: 
python3 --version

what is the answer?

cyndiac, 2020-07-14:
Python 3.7.3

AaronYoung5, 2020-07-14:
Hi Cyndia,

I build from PyChrono from source on my mac. I ran into a similar issues as you. The fix for me was to explicitly set the PYTHON_INCLUDE_DIR and the PYTHON_LIBRARY flags in cmake. I go to <path to chrono>/chrono/build and this is the command I call when building:
```
cmake \
  -DCMAKE_BUILD_TYPE:STRING=Release \
  -DCMAKE_C_COMPILER=$(which clang) \
  -DCMAKE_CXX_COMPILER=$(which clang++) \
  -DENABLE_MODULE_PYTHON:BOOL=ON \
  -DPYTHON_EXECUTABLE:FILEPATH=/usr/local/bin/python3 \
  -DPYTHON_INCLUDE_DIR:FILEPATH=/usr/local/Frameworks/Python.framework/Versions/3.7/include/python3.7m \
  -DPYTHON_LIBRARY:FILEPATH=/usr/local/Frameworks/Python.framework/Versions/3.7/lib/libpython3.7.dylib \
  .. 
```
Alternatively, you can set these using ccmake.

Please replace the INCLUDE_DIR and LIBRARY paths that are actually on your system. Do you use brew? If so, it should automatically place your python installs at /usr/local/Frameworks/.... Let me know if this works!

cyndiac, 2020-07-17:
Does it matter if my python version is managed by conda? My file locations look like this:
 -DPYTHON_EXECUTABLE:FILEPATH=/Users/cyndiac/anaconda3/envs/bchrono/bin/python3 \
  -DPYTHON_INCLUDE_DIR:FILEPATH=/Users/cyndiac/anaconda3/envs/bchrono/include/python3.7m \
  -DPYTHON_LIBRARY:FILEPATH=/Users/cyndiac/anaconda3/envs/bchrono/lib/libpython3.7.dylib \
I also installed swig and eigen3 specifically in this conda environment. I didn't explicitly call these in ccmake - it already automatically filled these in.

One thing I noticed is that "which clang" outputs:
/usr/bin/clang
whereas "CMAKE_C_COMPILER" defaults to: "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc"
This isn't what you mentioned, but would this have any impact?

rainergericke, 2020-07-17:
python3.7 should be ok. I don't use anaconda, I prefer the build from source. My experience is, that it is better to stay away from the python versions which come with Xcode. I can recommend to load the latest python version from python.org. On my machine I have actually version 3.8.2 installed. It was the first python release which didn't cause any problems.

cyndiac, 2020-07-21:
Just to be clear, I was still trying to build from source, but I used anaconda to manage the python packages that are called (i.e. python 3.7, swig, eigen3). Aaron's suggestion didn't seem to make a difference since cmake was already defaulting my python directories to the ones within my conda environment. 

I deactivated anaconda and used the brew install version of python (3.8.3), following Aaron's cmake command, and I got this error upon `import pychrono`:
```
Cyndias-MacBook-Pro:demos cyndiac$ python demo_python_1.py
First tutorial for PyChrono: vectors, matrices etc.
Traceback (most recent call last):
  File "demo_python_1.py", line 15, in <module>
    import pychrono as chrono
  File "/Users/cyndiac/Documents/GitHub/chrono_build/bin/pychrono/__init__.py", line 37, in <module>
    from .core import *
  File "/Users/cyndiac/Documents/GitHub/chrono_build/bin/pychrono/core.py", line 13, in <module>
    from . import _core
ImportError: cannot import name _core
```
or
`ImportError: dynamic module does not define init function (init_core)`

AaronYoung5, 2020-07-21:
Hi Cyndia,

Sorry about the late response. If you can, could you try `python3 demo_python_1.py` instead of `python demo_python_1.py`? If you haven't removed it manually, you should still have python2.7 on your system (it comes on Mac's by default), which is what regular `python` references.

Let me know if that works. Otherwise, I may have found a fix with building from source within conda, but I haven't gotten the chance to thoroughly test it.

cyndiac, 2020-07-22:
Thanks Aaron. Deactivating conda and making sure to call `python3` stopped the import segmentation fault problem.

closed 2020-11-13 by rserban