projectchrono/chrono #192: issue with irrlicht - black screen

ZhiquanW, 2019-10-23:
When i try to run codes with IrrApp in while loop like:
```
while True:
    irr_app = irr.ChIrrApp(system, 'irr', irr.dimension2du(800, 600))
    ...
    while irr_app.GetDevice().run():
        irr_app.BeginScene()
        irr_app.DrawAll()
        system.DoStepDynamics(0.01)
        irr_app.EndScene()
        if sim_env.GetChTime() > 10:
            irr_app.GetDevice().closeDevice()
            break
```
The rendering is correct at the first time(the first loop), and turn black in the following loops.

rserban, 2019-10-23:
I'm not sure what sim_env is here.  Are you sure you didn't mean to use `if system.GetChTime() > 10:`?

ZhiquanW, 2019-10-23:
Sorry , i forget to change the name to system:ChSystemNSC()

ZhiquanW, 2019-10-23:
What i want to do is restart the simulation after 10 seconds.