projectchrono/chrono #193: Error building irrlicht demos on Ubuntu

barf, 2019-10-24:
Hello, I am trying to build the irrlicht demos on Ubuntu, but there is an error saying `‘_IRR_IMPLEMENT_MANAGED_MARSHALLING_BUGFIX’ was not declared in this scope`.

Output:
`[ 41%] Building CXX object src/chrono_irrlicht/CMakeFiles/ChronoEngine_irrlicht.dir/ChIrrCamera.cpp.o
/home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.cpp: In member function ‘virtual bool chrono::irrlicht::RTSCamera::isInputReceiverEnabled() const’:
/home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.cpp:199:5: error: ‘_IRR_IMPLEMENT_MANAGED_MARSHALLING_BUGFIX’ was not declared in this scope
  199 |     _IRR_IMPLEMENT_MANAGED_MARSHALLING_BUGFIX;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.cpp: In member function ‘void chrono::irrlicht::RTSCamera::recalculateViewArea()’:
/home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.cpp:281:32: error: no matching function for call to ‘irr::scene::SViewFrustum::SViewFrustum(irr::core::matrix4&)’
  281 |     ViewArea = SViewFrustum(mat);
      |                                ^
In file included from /usr/local/include/irrlicht/irrlicht.h:190,
                 from /home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.h:16,
                 from /home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.cpp:13:
/usr/local/include/irrlicht/SViewFrustum.h:164:9: note: candidate: ‘irr::scene::SViewFrustum::SViewFrustum(const matrix4&, bool)’
  164 |  inline SViewFrustum::SViewFrustum(const core::matrix4& mat, bool zClipFromZero)
      |         ^~~~~~~~~~~~
/usr/local/include/irrlicht/SViewFrustum.h:164:9: note:   candidate expects 2 arguments, 1 provided
/usr/local/include/irrlicht/SViewFrustum.h:147:9: note: candidate: ‘irr::scene::SViewFrustum::SViewFrustum(const irr::scene::SViewFrustum&)’
  147 |  inline SViewFrustum::SViewFrustum(const SViewFrustum& other)
      |         ^~~~~~~~~~~~
/usr/local/include/irrlicht/SViewFrustum.h:147:56: note:   no known conversion for argument 1 from ‘irr::core::matrix4’ {aka ‘irr::core::CMatrix4<float>’} to ‘const irr::scene::SViewFrustum&’
  147 |  inline SViewFrustum::SViewFrustum(const SViewFrustum& other)
      |                                    ~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from /usr/local/include/irrlicht/irrlicht.h:190,
                 from /home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.h:16,
                 from /home/barf/chrono/src/chrono_irrlicht/ChIrrCamera.cpp:13:
/usr/local/include/irrlicht/SViewFrustum.h:48:3: note: candidate: ‘irr::scene::SViewFrustum::SViewFrustum()’
   48 |   SViewFrustum() : BoundingRadius(0.f), FarNearDistance(0.f) {}
      |   ^~~~~~~~~~~~
/usr/local/include/irrlicht/SViewFrustum.h:48:3: note:   candidate expects 0 arguments, 1 provided
make[2]: *** [src/chrono_irrlicht/CMakeFiles/ChronoEngine_irrlicht.dir/build.make:128: src/chrono_irrlicht/CMakeFiles/ChronoEngine_irrlicht.dir/ChIrrCamera.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1377: src/chrono_irrlicht/CMakeFiles/ChronoEngine_irrlicht.dir/all] Error 2
make: *** [Makefile:163: all] Error 2`

Is this a problem with my build environment? Or is it some bug ?

barf, 2019-10-25:
I used the main subversion trunk, the Chrono CMake configuration found it in /usr/local/include
https://svn.code.sf.net/p/irrlicht/code/trunk

rserban, 2019-10-25:
Indeed, it looks like version 1.9 will have some API changes.
Please use one of the stable releases. 1.8.4 works fine.

barf, 2019-10-25:
just to answer my own question, building against irrlicht-1.8.4 seems to work! yay!
but just FYI apparently irrlicht 1.9 removes something which breaks things