projectchrono/chrono #194: conda-forge package

davidbrochart, 2019-10-24:
I am trying to create a package on conda-forge: https://github.com/conda-forge/staged-recipes/pull/9965
But the CI fails with conflicts. Has anyone tried to to that?

Benatti1991, 2019-10-24:
Hi David,
we are already building conda packages (they we are uploading them to project chrono channel); this being said, in the contrib/packaging_python/conda directory you can find conda recipes. 
BTW, I noticed that you have "ENABLE_MODULE_PYTHON=OFF", you should turn it ON in order to build the python module. 
This being said, don't hesitate to directly contact me, I'll be happy to help you in building the conda-forge package!
Simone

davidbrochart, 2019-10-24:
Hi Simone,
Thanks for your answer. I saw the chrono channel, but I would like to have it in the conda-forge channel.
I had to set "ENABLE_MODULE_PYTHON=OFF" because of compilation issues. But now that I have enabled it I get conflicts:
https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=83937&view=logs

Benatti1991, 2019-10-25:
Hi David,
I've seen your last build (https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=84142) and it is failing because it does not find Python.h. Make sure that python.h is in the directory pointed by "DPYTHON_INCLUDE_DIR"
Simone

davidbrochart, 2019-10-25:
Thanks for taking a look Simone. I had try other things, now it fails with error "directorout typemap for plain pointer not implemented":
https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=84166

Benatti1991, 2019-10-25:
Oh, I remember it. The code in master branch is not compatible with SWIG 4.0. You could:
- build the code in the develop branch (which I suggest)
- Downgrade swig to 3..012

davidbrochart, 2019-10-25:
Thanks Simone, downgrading swig worked. `conda-forge` needs a release so taking the develop branch is not an option.

davidbrochart, 2019-10-27:
Actually, it doesn't install the Python package: https://github.com/conda-forge/staged-recipes/pull/9965
The feedstock is [here](https://github.com/conda-forge/pychrono-feedstock), so you can `conda install -c conda-forge pychrono`, but then `import pychrono` doesn't work. Any idea?

Benatti1991, 2019-10-28:
HI David,
for some reason the package is in the wrong directory. I found it in 
`/home/user_name/miniconda3/envs/env_name/share/chrono/python/pychrono`
Instead of 
`/home/user_name/miniconda3/envs/env_name/lib/python_ver/site-packages/pychrono`

davidbrochart, 2019-10-28:
Thanks Simone, it works if I copy the package in `site-package`, but only for Linux. For OS X I have a `illegal instruction` error when importing pychrono. Do you know where that can come from?

Benatti1991, 2019-10-28:
We don't, we are working to fix the issue on MasOS, but currently I don't have access to a Mac machine so it's quite difficult for me to fix the problem. Are you experiencing the same issue mentioned [here](https://github.com/projectchrono/chrono/issues/191)?

davidbrochart, 2019-10-28:
It could be related but I'm not sure. I don't have a Mac machine either, I just saw the error in the CI, which makes it difficult to debug too. I disabled OS X for now. Thanks for your answer!

davidbrochart, 2019-11-18:
Hi Simone, any news about the OSX issue? Here is the error in the CI: https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=95184

Benatti1991, 2019-11-19:
I don't know what's in test.py line 2, but I assume it is "import pychrono". We are experiencing this issue and I couldn't get to work on a Mac machine to inspect it, sorry.

Benatti1991, 2019-12-09:
Hi David,
we are trying to solve the issue with MacOS. 
Could you please tell me which compiler and SWIG version you are using to build PyChrono for Mac?
Thanks,
Simone

davidbrochart, 2019-12-09:
Hi Simone,
We use `swig 3.0.12` (see the [recipe](https://github.com/conda-forge/pychrono-feedstock/blob/master/recipe/meta.yaml)). The compilers in the [CI](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=95184) are:
```
+CC=x86_64-apple-darwin13.4.0-clang
+CXX=x86_64-apple-darwin13.4.0-clang++
```
Thanks for working on it!

rainergericke, 2019-12-11:
Hi David,

we were successful building chrono with python enabled. Actually we cannot support mkl and cascade and every SolidWorks related stuff. Unfortunately I cannot help you building Conda packages because I don't have any idea about it. I use homebrew as package manager, but when you see what I did with homebrew, you can imagine what to do with Conda hopefully.

brew install cmake doxygen graphviz gnuplot eigen python swig
brew install glm glew glfw ninja irrlicht ffmpeg

homebrew installs python-3.7.5 and swig-4.0.1. That seems to be a good combination, but we have to consider that python2 is still present and we have to be sure to point to python3 whenever necessary in the cmake calls. It should be possible to build chrono with python support now. Command line tools like demo_python_1.py must be started with

python3 demo_python_1.py

The most python examples need Irrlicht for 3D graphics, they have to be wrapped into a Mac application bundle. There is a nicer helper available: py2app it must be installed with

pip3 install py2app

Good Luck

Rainer

Benatti1991, 2019-12-11:
Rainer,
thank you again for your help. I'm adding a couple of consideration related to this particular issue:

- Python and pip version are determined by Anaconda environment, so the right ones are picked given that the environment is activated.
- SWIG 4 seems to be the solution to our problem but since you are building the code from the master branch it would not compile because chrono 4.0 release isn't compatible with SWIG 4. Good news: chrono 5 is likely to be released soon.
Simone

davidbrochart, 2019-12-11:
Thanks Rainer and Simone. I guess it's best to wait until chrono 5 is released then. Thanks again!

rserban, 2019-12-11:
David,
The 5.0 release will only happen end of January, mid-February (if we get to finish up everything we need for it).  Also, it would be good to know **before** the release if this also works for your setup.  So far we have only Rainer's experience as the unique data point.
Long story short, could you maybe try this out with the latest code in the 'develop branch? And let us know how it goes?
--Radu

davidbrochart, 2019-12-11:
Hi Radu, the thing is that conda-forge requires a release, and forbids to take just a branch. Could you please make a "dummy" release, on the develop branch?

rserban, 2019-12-11:
What constitutes a "dummy" release?  Will a tag be enough? If so, any particular name for it?

davidbrochart, 2019-12-11:
Actually, let me try without a release, I might be able to download a tarball through `https://github.com/projectchrono/chrono/tarball/develop`.

davidbrochart, 2019-12-12:
@rserban I tried to disable irrlicht for now with `-DENABLE_MODULE_IRRLICHT:BOOL=OFF`, but I'm still getting an error that seems related to irrlicht:
```
[ 99%] Linking CXX shared library ../../../lib/libChronoModels_vehicle.so
[ 99%] Built target ChronoModels_vehicle
Scanning dependencies of target vehicle_swig_compilation
[100%] Swig compile ChModuleVehicle.i for python
/home/conda/feedstock_root/build_artifacts/pychrono_1576113474141/work/src/chrono_python/ChDriver.i:34: Error: Unable to find 'dimension2d.h'
```

rserban, 2019-12-12:
You're right, the vehicle module wrapper should conditionally include irrlicht-related stuff.  We'll look into fixing this.   In the meantime, you should also disable the vehicle module (ENABLE_MODULE_VEHICLE).

rserban, 2019-12-12:
@davidbrochart I pushed a fix to allow building the python wrapper for Chrono::Vehicle even with Irrlicht disabled.   Please let us know if you get this working on Mac

rainergericke, 2019-12-12:
I could build it on the Mac, we don't have 'text mode' vehicle example in python yet. The demo_python_*.py examples work.

rainergericke, 2019-12-12:
New problem on the Mac. When I build chrono with vehicle and Irrlicht the build process works.
The python vehicle examples run into an error:
AttributeError: module 'pychrono.vehicle' has no attribute 'ChWheeledVehicleIrrApp'
The other Irrlicht based examples work.

rserban, 2019-12-12:
Yeah, looks like my "fix" broke things.

rserban, 2019-12-12:
Rainer, I pushed a fix to the fix.

davidbrochart, 2019-12-12:
I still get a `illegal instruction` error: https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=103125

rainergericke, 2019-12-13:
Things seem to be functional on the Mac now. py_chrono can be configured with or without Irrlicht.

Benatti1991, 2019-12-16:
Hi David,
we forgot to mention that we were able to build and import pychrono using CLang 11.
Would  it be possible for you to update to that version?
Thanks,
Simone

davidbrochart, 2019-12-23:
Hi Simone, sorry for the late reply. In conda-forge, the version of clang is 9.0.0, and I don't think we can use a newer version. It would be nice if you could check that it doesn't work with 9.0.0 on your side, so that we know that the clang version is indeed the issue.

davidbrochart, 2019-12-23:
Actually, what do you mean by clang 11? This version of clang doesn't exist yet.

rserban, 2019-12-24:
I'm not an OSX user, but apparently Apple re-versions clang to match the XCode version.  
According to https://en.wikipedia.org/wiki/Xcode#Xcode_7.0_-_11.x_(since_Free_On-Device_Development), Xcode 11.3 (Apple clang version 11.0.0) seems to be based on LLVM 8.0.0.

rainergericke, 2019-12-24:
Radu is right. Apple has its own naming scheme. When we talk about clang versions we should also mention which compiler we mean. The llvm compilers can also be built on the Mac and they are able to build chrono. It is also possible that not every Mac user uses the latest MacOS and/or the latest Xcode release, so David may be also right. This is my configuration:
MacOS 10.15.2 Catalina
clang Apple clang version 11.0.0 (clang-1100.0.33.16)
Xcode 11.3

davidbrochart, 2019-12-24:
The problem is that on conda-forge we have strict requirements on compiler versions. Today we use clang 9.0.0. It would be nice to check that on your side it doesn't work with clang 9.0.0.

rainergericke, 2019-12-24:
Which answer do you get with this command:
$ clang --version

davidbrochart, 2019-12-24:
I can't do that easily because it runs on the CI, but I can see in the log that it's 9.0.0.

rainergericke, 2019-12-24:
It would be good to know, because once upon a time also Apple clang-9.0.0 existed and it is quite far away from LLVM clang-9.0.0. I also have llvm-clang on my Mac and I can give it a try.

davidbrochart, 2019-12-24:
I was able to run the command: `clang version 9.0.0 (tags/RELEASE_900/final)`

rainergericke, 2019-12-24:
It looks like LLVM. I will try it.

rainergericke, 2019-12-24:
PyChrono + LLVM-clang + SWIG-4.0.1 + python3 works.

davidbrochart, 2019-12-24:
So now we need to compare our environments. The compilation log on the CI is here: https://dev.azure.com/conda-forge/84710dde-1620-425b-80d0-4cf5baca359d/_apis/build/builds/103125/logs/20
Could you please send me yours?

rainergericke, 2019-12-24:
Before I send my log file, just a remark, your compiler has been built in the environment
x86_64-apple-darwin13.4.0-clang. I use x86_64-apple-darwin19.2.0 for Apple-clang and for LLVM-clang. That means you are working with an older Xcode/MacOS. But it doesn't matter - it seems to me your package builds well and complete as configured. The problem might be, that the environment variable PYTHONPATH is not set correctly in the test situation. Cmake gives an information how it has to be set. There are two possibilities: one for runing in the build directory and an other for running in the install directory.

davidbrochart, 2019-12-24:
But conda does't make any use of the PYTHONPATH environment variable. Also, after installation we `cp -r $PREFIX/share/chrono/python/* $PREFIX/lib/python$HOST_PY_VER/site-packages`. I am trying to test with `python -c "import os, site; os.environ['PYTHONPATH'] = site.getsitepackages()[0]; import pychrono"`. However, even if that works, we won't be able to do that in the user's environment.

Benatti1991, 2019-12-24:
The pythonpath variable is not likely to be the issue. The interpreter finds pychrono, otherwise you would get a different error.

davidbrochart, 2019-12-24:
Indeed it didn't fix the issue.
@isuruf mentioned that we should not link to libpython3.8.dylib, because the python executable is statically linked to the python library, so if an extension is linked to the shared library, it'll bring in 2 copies.
Is there a way to not link to the python library currently?

davidbrochart, 2019-12-25:
Also, there quite a lot of compiler warnings in OSX (https://dev.azure.com/conda-forge/84710dde-1620-425b-80d0-4cf5baca359d/_apis/build/builds/106163/logs/20) that we don't have in Linux (https://dev.azure.com/conda-forge/84710dde-1620-425b-80d0-4cf5baca359d/_apis/build/builds/106163/logs/16).

rainergericke, 2019-12-25:
If you are able to build the whole package, the warnings don't matter. The end product will work. Seems the clang compilers are more penible and verbose than others. Does the c++ stuff work?

davidbrochart, 2019-12-25:
Not sure about the warnings being harmless: https://stackoverflow.com/questions/14268887/what-is-the-illegal-instruction-4-error-and-why-does-mmacosx-version-min-10
I don't know about the C++, how can I test it?

rainergericke, 2019-12-25:
Depending on the cmake configuration some tests and/or demos are built. The best way would be to activate the unit tests. One of the basic demos is 'demo_CH_EulerAngles.app'. Most c++ demos are built as MacOS application bundles. To start a bundle from the command line you must write:
$ demo_CH_EulerAngles.app/Content/MacOS/demo_CH_EulerAngles
You must go to the directory where the bundles sits before you can start it.
When you have unit tests activated, you have programs with name beginning with utest....
you can start them as normal programs you know from Linux.

cekees, 2020-02-05:
Hi @rserban, is there any chance we'll get a new release soon? I believe the issue is fixed, but the conda forge package is waiting on a new chrono release. Let us know if we can help test. /cc @davidbrochart .

davidbrochart, 2020-02-05:
@cekees I'm not sure we are talking about the same issue. This issue has to do with pychrono failing in the CI on OSX. AFAIK this has not been fixed yet.

rserban, 2020-02-05:
@cekees We are planning on a release by the end of the month.

cekees, 2020-02-05:
OK, thanks. @davidbrochart I see. Can you enable the osx version of the conda build and just allow the warnings so @ejtovar can test on osx? I tend to agree with @rainergericke that  disabling the demos and enabling the unit tests in the conda pacakge would be a good way forward. I don't think the demos should be built in the conda package. To be clear, my goal here is to get the chrono conda package functional for both linux and osx.

davidbrochart, 2020-02-06:
> OK, thanks. @davidbrochart I see. Can you enable the osx version of the conda build and just allow the warnings so @ejtovar can test on osx?
I don't see what you mean, there is [this PR](https://github.com/conda-forge/pychrono-feedstock/pull/6) where we enable the OSX build. The test fails so this will not help @ejtovar test.

cekees, 2020-02-06:
I meant enable the osx build of chrono on conda-forge. I do see that your conda-forge chrono feedstock PR to enable OSX is failing to build on OSX, but I don't understand the issues.  I gather from comments above that @rainergericke and @Benatti1991 have functioning osx builds.  Should @ejtovar just get chrono from the chrono channel mentioned above or homebrew for now? @davidbrochart I can't tell from comments above whether you were ever able to build a pre-release version of chrono or not on osx.  Can you confirm that  the upcoming release of chrono is going to fix the conda-forge package?

davidbrochart, 2020-02-06:
@cekees I have never been able to get pychrono to work on OSX in conda-forge. When we import pychrono we get a `Illegal instruction` error. @rainergericke and @Benatti1991 have functioning OSX builds *locally*, but we need it in the CI. The upcoming release is likely not to fix the conda-forge package because the PR takes the `develop` branch and the release will be based on it.
I don't advise that @ejtovar get pychrono from another channel because it can lead to binary inconsistency. I thought he was working on a workaround in Proteus that would enable OSX by disabling pychrono just for OSX?

Benatti1991, 2020-02-06:
This thread is becoming a bit confusing so, @davidbrochart , how does your situation look like now? Can you build PyChrono on the CI? Can you import? If not, what is the error that is thrown?
Also, do you want us to do something (test or whatever)?
Thanks,
Simone

[27 more replies]

closed 2020-02-25 by davidbrochart