projectchrono/chrono #195: Customize Torque Function In Python

ZhiquanW, 2019-10-29:
Here is my codes:
```
class LegMotor(chrono.ChFunction):
    def __init__(self):
        chrono.ChFunction.__init__(self)
    def Get_y(self, x):
        return 1.0
...
f = LegMotor()
tmp_motor.SetTorqueFunction(f)
```

The error message is:
```
Traceback (most recent call last):
  File "main.py", line 94, in <module>
    sim_env.DoStepDynamics(0.01)
  File "/home/zhiquan/anaconda3/envs/dev/lib/python3.7/site-packages/pychrono/core.py", line 16530, in DoStepDynamics
    return _core.ChSystem_DoStepDynamics(self, m_step)
RuntimeError: SWIG director method error. Error detected when calling 'ChFunction.Update'
```

Benatti1991, 2019-10-29:
Hi Zhiquan,
I cannot reproduce your error, so please try installing the latest develop version and then report me if you are still experiencing this issue.
To do so, in your conda env, type:
`conda install -c projectchrono/label/develop pychrono`

Benatti1991, 2019-10-30:
It is possible, but I cannot tell what you are doing wrong. When the swig generated codes invokes an overridden method the error message is generic (and usually due to some mistake in the overridden method). Could you please paste here the whole code, or alternatively send it privately to me so that I can reproduce your error?

ZhiquanW, 2019-10-30:
Thank you for your help.
Here is the link of my repository:
https://github.com/ZhiquanW/Species-Evolution-Simulator
The related codes are:
ChronoSim/Creature.py : Line6-11 / Line 94-109

ZhiquanW, 2019-10-30:
I can uncomment  ChronoSim/Creature.py Line 105 & comment Line 106 to make it run successfully

ZhiquanW, 2019-10-31:
Thank you for your help!!!
Before I implemented this customized function, i searched for examples about that in demo_IRR_motors.cpp. I did not realize that i need to keep them during the simulation instead of  using them as a temporary value like what example4.3 do.

closed 2019-11-04 by Benatti1991