projectchrono/chrono #198: Pychrono import error

5Volts, 2019-11-11:
I've installed Chrono with [this](http://api.projectchrono.org/development/tutorial_install_chrono.html) guide. I then proceed to install pychrono module using [this](http://api.projectchrono.org/development/module_python_installation.html) (Note, the process intertwines with the installing Chrono, they are not separate). I've also installed Anaconda and use that to install pychrono library. After activating the conda environment, I tried importing the pychrono library but no luck.

```
C:\Users\STUDENT\Desktop\New folder (2)>conda activate base
(base) C:\Users\STUDENT\Desktop\New folder (2)>py
Python 3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pychrono
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\STUDENT\Anaconda3\lib\site-packages\pychrono\__init__.py", line 7, in <module>
    from .core import *
  File "C:\Users\STUDENT\Anaconda3\lib\site-packages\pychrono\core.py", line 13, in <module>
    from . import _core
ImportError: DLL load failed: A dynamic link library (DLL) initialization routine failed.
>>>                                                                                           
```
I googled the error but all that came up is TensorFlow library, not pychrono.

Benatti1991, 2019-11-11:
You mentioned that you followed the instructions to build from sources, but it looks like your your pychrono module is in an Anaconda directory.  It looks like your installation needs some cleanup, I think you installed pychrono with anaconda and Conda still finda that one. 
Moreover, it is installed in the base environment, which I usually do **not** recommend. 
Simone

5Volts, 2019-11-11:
Is there a way I can install it with pip? I am not familiar with anaconda.

Benatti1991, 2019-11-11:
Currently there's no pip package available for pychrono.
The Anaconda installation guide is here:

http://api.projectchrono.org/development/pychrono_installation.html

To manage Anaconda environments please refer to this:

https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html

5Volts, 2019-11-11:
I've followed all the instructions. I've even uninstalled python and anaconda, and reinstall it back but still the same error.

What cleanup is necessary here?

Benatti1991, 2019-11-11:
Have you installed it in a separate environment? Which version have you installed? Develop or main? If you follow the instructions you install the main (which is totally OK )

5Volts, 2019-11-11:
I installed it in the base environment and following the instructions involves downloading this repository, and when it is downloaded, it is named chrono-develop, so it is probably the develop version.

Where can I find the main version?

Benatti1991, 2019-11-19:
Please do not install it in the base environment. If you don't specify label/develop during conda install you will get the main version (that was develop when it was built actually)

5Volts, 2019-11-19:
I plan on using some other simpler physics engine, in the future when I have a Linux machine, I'll come back for this.

closed 2019-11-19 by 5Volts