projectchrono/chrono #199: Import error

LeoG2, 2019-11-29:
hi
I have been trying to install pychrono. but when I  try to load it appears the following : 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Admin\Anaconda3\envs\shapeSim\lib\site-packages\pychrono\__init__.py", line 7, in <module>
    from .core import *
  File "C:\Users\Admin\Anaconda3\envs\shapeSim\lib\site-packages\pychrono\core.py", line 13, in <module>
    from . import _core
ImportError: DLL load failed: A dynamic link library (DLL) initialization routine failed.

I set up a new environment , nenv, and used conda install -c projectchrono pychrono. 
the system states as done. Using list I can see pychrono. 

pychrono                  4.0.0                  py37_402    projectchrono

Does anyone could please help me to solve this issue? what is required to get pychrono running.

Benatti1991, 2019-11-30:
Hi Leo,
does your CPU support AVX2 instructions?

LeoG2, 2019-11-30:
Hi ,
yes it does, it is an intel core i5 8400T.

LeoG2, 2019-12-02:
> Hi ,
> yes it does, it is an intel core i5 8400T.

Hi Benatti, 
Sorry, I have double checked, and the processor only supports AVX instructions. The issue started after updating the libraries. I installed using anaconda and I manage to run some codes importing some obj files. The test were random as sometimes the files could be uploaded and showed on the screen, but using different files with the same extension did not work. I also tried to load stl files, which worked fine the first time, but the following day stop working using the same file and any other file, not loading the file. Then I updated pychrono and anaconda to load them as STEP files with cascade, but the library was not installed. The issue started after the update.

Now when I tried to import pychrono I have the next message: 

import pychrono
Traceback (most recent call last):

  File "<ipython-input-3-976e2f1da79d>", line 1, in <module>
    import pychrono

  File "C:\Users\Admin\Anaconda3\envs\shapeSim\lib\site-packages\pychrono\__init__.py", line 7, in <module>
    from .core import *

  File "C:\Users\Admin\Anaconda3\envs\shapeSim\lib\site-packages\pychrono\core.py", line 13, in <module>
    from . import _core

ImportError: DLL load failed: A dynamic link library (DLL) initialization routine failed.

guosijia, 2022-04-25:
I have the same problem, is there any way to solve it?

uebian, 2025-11-26:
Thank you for raising the issue. Our conda package building pipeline has recently changed and we haven't updated the document. We have simplified the installation process so you won't need to manually download and install packages as before. Could you please try using the following two command in the NEW environment to install PyChrono through conda? Thank you very much!

```
conda create -n chrono_test python=3.12 -c conda-forge
conda install projectchrono::pychrono -c conda-forge
```