projectchrono/chrono #200: Core dumped error [Linux]

Hugi21, 2019-12-03:
Hello ! 
I recently installed pyChrono and I tried to run that code to test if everything works : http://api.projectchrono.org/tutorial_pychrono_demo_python2.html
But I have a core dumped error when I try to run it. (I don't put the logs because visual studio just display "Core dumped").

I saw that there were issue about that error but all the library is correctly installed. So I don't know what I can do to solve that.

Here is the result of `$ more /proc/cpuinfo | grep flags` : 

`flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer xsave avx f16c lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm arat pln pts md_clear flush_l1d`

So if you can help me :).

Regards

Benatti1991, 2019-12-06:
Hi Hugi,
thanks a lot for posting your CPU flags! =) Indeed, the conda binaries need AVX2. 
You can build pychrono  from sources following these instructions:
- Building the core and irrlicht module:
http://api.projectchrono.org/development/tutorial_install_chrono.html
- Building the Python Module (PyChrono):
http://api.projectchrono.org/development/module_python_installation.html

AlexandrParkhomenko, 2020-03-14:
Hello friends! 
I build pychrono from sources:
Program received signal SIGILL, Illegal instruction.
0x00007ffff77700e3 in ?? () from /usr/local/lib/libChronoEngine.so

flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm 3dnowext 3dnow constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid pni monitor cx16 popcnt lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt hw_pstate vmmcall npt lbrv svm_lock nrip_save

I try debug now

uebian, 2026-01-14:
The released conda binary is intended to compile with newer CPU architecture for better performance. If your CPU doesn't support certain features, you might want to compile from sources. If you still encounter the issue even if you build from source, please open a new ticket.