projectchrono/chrono #206: FSI module

joeelrahi, 2019-12-20:
I cannot build the FSI module, the following error lists pops out:

[image]

note: other modules including IRRLICHT work perfectly.

Do you have any idea what could be the problem?

rserban, 2019-12-20:
There are many possible reasons for this error in Visual Studio.  You can find out more looking at the 'Output' window instead of the 'Error List' window (see https://stackoverflow.com/questions/13118947/error-msb6006-cmd-exe-exited-with-code-1)

Since you mention that other modules build fine, my first suspicion is that this is related to CUDA.  
- is CUDA found properly during CMake configuration? Check the CMake output which will report the CUDA version, path to the CUDA toolkit, results from automatic GPU detection, etc.
- what versions of VS and CUDA do you use?  Not all combinations are compatible.
- can you successfully build a CUDA test program?

Note also that, in order for our CMake scripts to correctly identify the GPU and set the appropriate CUDA compile flags, your PATH environment variable must contain the path to cl.exe

joeelrahi, 2019-12-20:
Hello,

Thanks for your prompt response! I have CUDA 9.2 and VS 2019.

I checked the cmake configuration and it finds the CUDA but fails to detect the GPU

""Searching for CUDA...
  CUDA version:          9.2
  CUDA toolkit root dir: C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA/v9.2
Automatic GPU detection failed. Building for all known architectures.
Compiling for CUDA architecture: 3.0 3.5 5.0 6.0 6.1 7.0 7.5""

I will try to downgrade to CUDA 7.5 and give it another try.

rserban, 2019-12-20:
Before that, I suggest you try to fix the GPU detection issue by adding to your PATH environment variable the path to cl.exe.

closed 2019-12-20 by joeelrahi

rserban, 2019-12-20:
I take it this solved the issue?

joeelrahi, 2020-01-03:
Hello again, 

yes seems that the problem is because of the GPU on my computer. Anyway, regarding the demos: seems like the Flexbile Shell (demo_FSI_Flexible_Shell.cpp) is missing, would you be able to share the cpp file with me (standalone is okey, I don't wana run it just have a look at the script).

kind regards,
Joe

rserban, 2020-01-03:
I may have a solution to the automatic detection of the GPU that would not require modifying the PATH environment variable.  I need to do a bit more testing before pushing a fix.

As for that demo, it is part of the distribution: https://github.com/projectchrono/chrono/blob/develop/src/demos/fsi/demo_FSI_Flexible_Shell.cpp
Do you not see it in src/demos/fsi/ next to the other Chrono::FSI demos?

joeelrahi, 2020-01-03:
thanks a lot!