projectchrono/chrono #210: How to compile chrono under ubuntu16.04？

yumianhuli2, 2020-01-06:
Hello！
I tried to compile the chrono project on ubuntu16.04, I created the build folder in the root directory and then typed the command cmake ..
but terminal showed:
--   No NEON support found
-- Could NOT find Eigen3 (missing:  EIGEN3_INCLUDE_DIR EIGEN3_VERSION_OK) (Required is at least version "3.3.0")
CMake Error at src/CMakeLists.txt:218 (message):
  Eigen3 not found.  Specify EIGEN3_INCLUDE_DIR
-- Configuring incomplete, errors occurred!

So,How to compile chrono under ubuntu？Is there a tutorial for linux user？

rserban, 2020-01-06:
I believe the error message is self-explanatory:  you need to first get Eigen3 (version 3.3.0 or better, preferably the latest 3.3.7) and then, during CMake configuration of Chrono, specify the location of the Eigen3 headers by properly setting the variable EIGEN3_INCLUDE_DIR.

Installation instructions are available at http://api.projectchrono.org/tutorial_install_chrono.html

yumianhuli2, 2020-01-06:
> I believe the error message is self-explanatory: you need to first get Eigen3 (version 3.3.0 or better, preferably the latest 3.3.7) and then, during CMake configuration of Chrono, specify the [...]

ok Thank you！But this tutorial is for windows？！
[image]

rserban, 2020-01-06:
The installation instructions at the link I provided also apply to linux (os-specific issues are explicitly identified).
The issue may be with the old tool versions you use.  I do not have access to Ubuntu 16.04.  If you can not upgrade the os, maybe you can at least update your cmake version (which is now at 3.16) and maybe also gcc?
Having said that, some of the error messages you get are strange.  Could you maybe post your CMakeCache.txt file?

yumianhuli2, 2020-01-07:
ok!
successed？
[image]
[image]

rserban, 2020-01-07:
Yes, you successfully built the core Chrono module.  You may want to enable other Chrono modules for additional functionality, first of all Chrono::Irrlicht for run-time visualization.  See the installation instructions at http://api.projectchrono.org/development/tutorial_table_of_content_install.html

closed 2020-01-07 by rserban