projectchrono/chrono #211: vehicle project setup  guide

yumianhuli2, 2020-01-07:
I found that https://github.com/projectchrono/chrono-tutorial, Where do I need to clone this demo project to Chrono, and then compile the whole project？

rserban, 2020-01-07:
See the installation instructions at http://api.projectchrono.org/development/tutorial_table_of_content_install.html
and how optional modules (including the vehicle module) are enabled during CMake configuration.

Once you enable both the Irrlicht and Vehicle modules, you will get many more demos, including several for various vehicle simulations using various different vehicle models.   I recommend starting with those.

The projects in the chrono-tutorial repository are set up as "external" projects using the Chrono libraries.  To see how to build such projects, see http://api.projectchrono.org/development/tutorial_install_project.html 

Various vehicle models have different numbers of DOFs.

I assume you ask about whether or not a single vehicle will run in (soft) real-time on a performant machine.  The answer is yes, it should.

Ray casting can be done using the underlying collision detection library.  This is an advanced topic, but look at the ChCollisionSystem::RayHit function in the API documentation.

closed 2020-01-07 by rserban

yumianhuli2, 2020-01-07:
> See the installation instructions at http://api.projectchrono.org/development/tutorial_table_of_content_install.html
> and how optional modules (including the vehicle module) are enabled during CMake [...]

Hello! I installed IRRLICHT1.8.4 and cmake-gui.But I got below:
[image]
[image]
[image]

rserban, 2020-01-07:
This is likely due to the fact that you copied that demo from the chrono-tutorial repository into your Chrono clone.    As I've already told you, projects such as those in 'chrono-tutorial' are built differently (see the link I provided).

Simply clean up your Chrono clone, clean the cmake cache, and reconfigure.

You should not see a variable 'Chrono_DIR' when configuring Chrono.