projectchrono/chrono #220: RigidTerrain intersection capped at 1000m

alhirzel, 2020-01-31:
When using a heightfield in excess of 1000m, the RigidTerrain::FindPoint function does not seem to work.

rserban, 2020-01-31:
Indeed, this value is hard-coded for the ray casting done in FindPoint.  I will look for a solution that does not use a hard-coded value.

alhirzel, 2020-01-31:
Assuming the terrain slope is always less than 90 degrees, I think you shoot two rays from (x,y,0) in the +z and -z directions.

rserban, 2020-01-31:
That would require twice the number of rays and will not always give the right result:  think of a flat horizontal box with its center at  z=100 and with height 10.  You would get a ray intersection point at a value z=95 when what you want is one at z=105.   Furthermore, for the ray casting you need to specify not only the start point for a ray, but also its end point.  Your approach would end up having to hard-code a "length" for these rays.

The solution I plan on implementing relies on calculating the AABB of all terrain collision shapes and then using its extent in the z direction to set the start and end points of the ray.

alhirzel, 2020-01-31:
Ah! Since the terrain admits general meshes you're right the double ray would not work.

rserban, 2020-02-01:
Let me know if the fix I pushed resolves your issue.

On a related note, if possible, try to stay away from collision shapes located very far from the global origin.  Indeed, collision detection is done in the global reference frame and you may experience precision issues.

commit 656d9fd4 (2020-02-01): Fixes to RigidTerrain:

- use a bounding sphere radius for each patch to set start and end points for ray-casting in FindPoint
- add option to set a sweeping sphere radius for mesh collision for height-field patches
- use a typed enum for the patch type

Resolves #220
closed 2020-02-01 by rserban