projectchrono/chrono #226: TypeError: in method 'ChSystem_SetSolver', argument 2 of type 'std::shared_ptr< chrono::ChSolver >'

by1e11, 2020-03-15:
Hi,

I am now using pychrono, but I encountered this error,
The code is really straightforward translation from the robot demo from the chrono c++

world = pch.ChSystemNSC()
world.SetSolver(pch.ChSolverPSOR)

I also suspect that this is a common error, as I tried to wrap the ChFunction_Setpoint and ChFunction_SetpointCallback. The wrapper is successfully generated, but when I was using it in python application, it thrown the same type of error message, something like
TypeError: in method 'ChLinkMotorLinearForce_SetForce', argument 2 of type 'std::shared_ptr<chrono::ChFunction_SetpointCallback>'

Could anyone can suggest wherever I might make mistake.
Thanks
Bo

Benatti1991, 2020-03-18:
Hi Bo,
SetSolver expects a (shared_ptr to) an instance of chrono::ChSolver, which you didn't create and  passed the definition instead. Try this:

```
world = pch.ChSystemNSC()
my_solver = pch.ChSolverPSOR()
world.SetSolver(my_solver)
```

Simone

by1e11, 2020-03-21:
> Hi Bo,
> SetSolver expects a (shared_ptr to) an instance of chrono::ChSolver, which you didn't create and  passed the definition instead. Try this:
> world = pch.ChSystemNSC()
> my_solver = pch.ChSolverPSOR()
> world.SetSolver(my_solver)
> 
> Simone

Thanks Simone, it seems that I made a stupid mistake. I will tried it out.

Bo

closed 2020-03-25 by Benatti1991