projectchrono/chrono #227: Use of std::make_unique

jpthalman, 2020-03-15:
Looking at the **5.0.1** release, there appears to be several uses of **std::make_unique**, which is a C++14 feature. Was breaking support for C++11 intentional?

rserban, 2020-03-15:
It was not intentional.  But all compilers we've tested have the C++14 support for this, so I decided not to roll this back.  
Is there a particular compiler you target that lacks support for std::make_unique?

jpthalman, 2020-03-15:
I believe we are using GCC 7.4 with the '-std=c++11' flag. The toolchain should be able to support make_unique, but I'm trying to integrate this into a large existing codebase which is limited to the C++11 standard.

closed 2020-03-18 by rserban

rserban, 2020-03-18:
@jpthalman I pushed a fix for this issue.  In your particular situation, since GCC 7.4 does actually support C++14, make sure to set the Chrono CMake variable CH_ENFORCE_CXX11 to true.

I replaced all instances of std::make_unique with a new function chrono_types::make_unique which currently falls back on using std::unique_ptr(new ...).

But you still need to set CH_ENFORCE_CXX11 to disable building a particular class (ChFunction_Lambda) that only works with C++14.

jpthalman, 2020-03-23:
Awesome! Thank you so much for looking into this so quickly. I'll pull the latest changes and test them out sometime next week.

commit 4987b8fb (2020-03-18): Allow strict C++11 standard:

- add function chrono_types::make_unique (since std::make_unique is only available in C++14, for now always use std::unique_ptr(new...)
- add CMake option CH_ENFORCE_CXX11 (default false) which forces C++11 (even if compiler supports C++14)

Resolves #227