projectchrono/chrono #229: I fixed a segfault

blacksph3re, 2020-04-07:
TL;DR: compile with `-DCMAKE_CXX_FLAGS='-std=c++11 -O2'`

As there are a couple of people here trying to fix segfaults, I also encountered segfaults when trying to get chrono 5.0.1 run on ubuntu. To be able to compile, I had to add --std=c++11, as libeigen uses some newer constructs since 3.3.6. I tried libeigen 3.3.2 to 3.3.7 and they all showed the same behavior:
```
Thread 1 "QBlade" received signal SIGSEGV, Segmentation fault.
0x00007f0119536198 in void Eigen::internal::pstore<double, double __vector(4)>(double*, double __vector(4) const&) ()
```
The crash happened upon loading of the library.

I eventually found [this issue](https://eigen.tuxfamily.org/bz/show_bug.cgi?id=1427) which tracked something similar it back to an alignment problem. Their fix was to compile their project with -O2, and though I can't explain why, this enabled me to run chrono on ubuntu. I hope this helps somebody!

rserban, 2020-04-09:
I cannot reproduce this issue.  Are you sure this is a Chrono problem and not with your code?  Do you get a segfault with one of the Chrono demos?

Also, how do you configure/build Chrono?  If you use the provided CMake scripts, --std=c++11 (or --std=c++14) should be added automatically (Chrono does require C++11).

What compiler do you use? What version?

blacksph3re, 2020-04-09:
Okay, you are probably right, it's most likely somewhere in my code, as the demos all run smoothly.
My code runs perfectly with the prebuilt chrono from [anaconda](https://anaconda.org/conda-forge/chrono), and also when I build libChronoEngine with clang. When I build with g++ 7.5.0, it segfaults.

The code I am running is a simulation, which I haven't coded myself - so I can not really tell much about the code. I will try to isolate the bug and come back to you. Thanks for your help so far

closed 2020-04-09 by blacksph3re