projectchrono/chrono #238: Python ImportError on MacOS from source install

nickgravish, 2020-05-27:
Hello, 

I followed the instructions for C++ API install with PyChrono. The install went fine. Now trying to import pychrono I get the error below. I am on MacOS 10.15.4, Python 3, using Anaconda but just installed from source.

```
Python 3.6.7 |Anaconda custom (64-bit)| (default, Oct 23 2018, 14:01:38) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.8.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pychrono as chrono                                                                                            
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-fc94b95448e4> in <module>
----> 1 import pychrono as chrono

~/chrono/chrono_build/bin/pychrono/__init__.py in <module>
     35 # instead of
     36 #   import pychrono.core
---> 37 from .core import *
     38 
     39 SetChronoDataPath('../../../data/')

~/chrono/chrono_build/bin/pychrono/core.py in <module>
     11 # Import the low-level C/C++ module
     12 if __package__ or "." in __name__:
---> 13     from . import _core
     14 else:
     15     import _core

ImportError: dynamic module does not define module export function (PyInit__core)
```

To get to this point I followed the tip in this issue (https://github.com/projectchrono/chrono/issues/172#issuecomment-533892293) to copy/paste the .so files into the package directory. Before I was getting an error that ```ImportError: cannot import name '_core```

Benatti1991, 2020-05-29:
Hi Nick,
as I said in the issue  you mentioned, that workaround is no longer needed, so there is something else not working. Could you please revert your copy&paste and report the full error message? 
Thanks,
Simone

nickgravish, 2020-05-29:
No problem, the error is as follows:

```
Python 3.6.7 |Anaconda custom (64-bit)| (default, Oct 23 2018, 14:01:38) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.8.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pychrono as chrono                                                                                            
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-fc94b95448e4> in <module>
----> 1 import pychrono as chrono

~/chrono/chrono_build/bin/pychrono/__init__.py in <module>
     35 # instead of
     36 #   import pychrono.core
---> 37 from .core import *
     38 
     39 SetChronoDataPath('../../../data/')

~/chrono/chrono_build/bin/pychrono/core.py in <module>
     11 # Import the low-level C/C++ module
     12 if __package__ or "." in __name__:
---> 13     from . import _core
     14 else:
     15     import _core

ImportError: cannot import name '_core'
```

Benatti1991, 2020-07-14:
Hi Nick, any update on this issue?
Simone

closed 2025-10-29 by uebian