projectchrono/chrono #240: segfault when using ChElasticityCosseratGeneric from pychrono (anaconda install)

albertoferna, 2020-06-04:
Hello,

I'm trying to set a simulation of a simple beam model for testing purposes. The geometry of the simulation is kind of irrelevant at this point as I'm just making a simple cantilever beam. What I'm interested in is the possibility of using a Cosserat beam model with my own 6x6 matrix.
Just by initializing an object of type ChElasticityCosseratGeneric in pychrono I get a segmentation fault. I'm using ubuntu 20.04. I have installed pychrono using conda.
I can create objects of type ChElasticityCosseratMesh, ChElasticityCosseratSimple, ChElasticityCosseratAdvanced, but not of type ChElasticityCosseratGeneric

Do you know what may be the issue?

tasora, 2020-06-05:
Dear Alberto,

  - probably it is an issue with the Python wrapping of the 
ChElasticityCosseratGeneric, because when I tested it in C++ it was 
fine. We will investigate the issue

- please send us the python source snippet where you create and use the 
ChElasticityCosseratGeneric, so that we can reproduce the issue

Last note: we are going to change the way that Gauss quadrature is done 
in the IGA beams for orders greater than 1, so for the moment please 
prefer order 1 (linear) for your developments, that won't be changed.

Alessandro Tasora

albertoferna, 2020-06-06:
Hello Alessandro,

Thank you for the information, I'll keep it in mind.
By the way, congratulations guys  for what looks like a very nice software.

I can reproduce the error in just 3 lines of code:
`import pychrono as chrono
import pychrono.fea as fea
melasticity = fea.ChElasticityCosseratGeneric()
`
I cannot thick why it would make a difference to create other objects before. But, as I'm not proficient with mixing languages in software (C++ and python in this case) and the mechanisms behind swig, I also tried a striped down version of the tutorial:
`
import pychrono as chrono
import pychrono.fea as fea

my_system = chrono.ChSystemNSC()

my_mesh = fea.ChMesh()
my_system.Add(my_mesh)

my_mesh.SetAutomaticGravity(True,1)
my_system.Set_G_acc(chrono.ChVectorD(0,-9.81, 0));

melasticity = fea.ChElasticityCosseratGeneric()
`
Looking at the code, it seems to me that ChElasticityCosseratGeneric() would be a correct constructor.

In case it helps I'm running the version installed from anaconda (conda install -c projectchrono pychrono) in ubuntu 20.04. I also tested it in two different computers (same install)

Thank you very much for your time

albertoferna, 2020-06-06:
Sorry I closed the issue unintentionally

tasora, 2020-06-12:
> I have some good news, I think.
> I have compiled the development version and the error is gone. I don't
> know exactly what has been changed in the code, but I can now create
> an object of type ChElasticityCosseratGeneric without getting a segfault.
> Now looking at the API I see no way of assigning matrix E, but that is
> a different question. Therefore, I'm closing this issue as it seems it
> is no longer applicable.

Hi Alberto,

we know the issue: there is no way (currently) in PyChrono to create 
arbitrary NxM fixed-size ChMatrixNM matrices because they rely on 
templating, which is not supported in Python. We just created the 
Pychrono equivalent for 3x3 case (the ChMatrix33D) but we are still 
thinking if it is worth implementing something like a ChMatrix66D just 
for passing it to the Cosserat elasticity... Probably a not elegant but 
more straightforward workaround is that we overload the c++ constructor 
of ChElasticityCosseratGeneric adding an alternative constructor that 
takes a ChMatrixDynamic type as argument, that would be wrapped easily 
in the Python side.

We will look at this in the next days.

regards

A. Tasora