projectchrono/chrono #249: Cascade missing from Conda Package

8bit-Dude, 2020-07-20:
Hey Guys, 
I installed pyChrono on Miniconda3, but noticed that the Cascade module is missing.
I wonder why, since pythonOCC 0.7.4 is also available on conda.

Anyhow, what would be the fastest way to get the cascade module installed? (The documentation implies that recompiling the whole lot with Cmake is required. That is not very appealing...)

Benatti1991, 2020-07-20:
Hi,
the conda package is available in the develop version of PyChrono. Please re-install using 
`conda install -c projectchrono/label/develop pychrono`
The support has been suspended due to some version conflict but now, as you correctly pointed out, every dependency is available on Anaconda.
Thanks,
Simone

8bit-Dude, 2020-07-20:
Thanks for the quick reply, I can confirm that moving to the development package fixed the issue! :-)

closed 2020-07-20 by 8bit-Dude