projectchrono/chrono #253: Issues related to Performance on Mac Catalina

CThuleHansen, 2020-07-30:
Hi.

I am experiencing very different performance from windows to mac when executing the same model:
https://github.com/INTO-CPS-Association/chronofmu/blob/ffo/Chrono_model/source/robotti_source.cpp

In windows, I can execute this with the arguments: `300 60 0 1` and it takes 2.46653 seconds.
On mac catalina 10.15.4, it takes 42.9905 seconds. 

I have checked the outputs of the model, and they are equal.

Mac gcc-version:
```                                                 
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
Ram: 16 GB

Windows visual studio 2019 and thereby MSCV v142 
INtel Core i7-5600U CPU @ 2.6 GHz 
RAM 12,0 GB

Is this something you are aware of or can help me pinpoint?

rainergericke, 2020-08-04:
Hi Thule,

I tried your program on my Mac with the arguments 300 60 0 1 and it took 
Simulation time: 0.91 -- end time: 1
Simulation time: 0.92 -- end time: 1
Simulation time: 0.93 -- end time: 1
Simulation time: 0.94 -- end time: 1
Simulation time: 0.95 -- end time: 1
Simulation time: 0.96 -- end time: 1
Simulation time: 0.97 -- end time: 1
Simulation time: 0.98 -- end time: 1
Simulation time: 0.99 -- end time: 1
Simulation time: 1 -- end time: 1
Simulation time: 1[s]
Real world time: 1.41602 [s]
Real/Sim ratio: 1.41602

which is pretty much like your 'windows time'.
Which Mac do you use?

Best
Rainer

CThuleHansen, 2020-08-05:
Hi Rainer, 
Thank you, that is very interesting.

My mac is:
# gcc-version:
```Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

# Specification
Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
Ram: 16 GB
[image]

Can you possibly upload the executable that you compiled so I can try and execute it on my machine and see how it behaves?

rainergericke, 2020-08-05:
Hi Casper,

Hier is my executable:

I think you have a good machine, it should deliver a similar performance like my iMac (3,2 GHz 6-Core Intel Core i7).
We will see….

Best

Rainer

CThuleHansen, 2020-08-05:
Hi @rainergericke I cannot see your executable, can you upload it to some file sharing site?

rainergericke, 2020-08-05:
Can you give me your mail address, please?

CThuleHansen, 2020-08-05:
casper.thule@eng.au.dk

CThuleHansen, 2020-08-05:
In case it is blocked by the email system, then just sent me a regular email and I can provide you a place to upload the executable via a file sharing site that Aarhus University recommends

CThuleHansen, 2020-08-05:
I have received it and will start changing the dylibs dependencies. I will let you know how it goes!

CThuleHansen, 2020-08-05:
Rainer, can you check whether you have some errors at the top of your execution - whether some obj files failed to load

CThuleHansen, 2020-08-06:
So thanks to @rainergericke assistance the issue is now fixed. I was building it in debug configuration.
**Thank you very much for you help.**

As a side note, that one might find interesting, here is a script that can change dependencies on dylibs, which is hardcoded when compiled.

I.e. when compiling a chrono application that uses libchronoEngine.dylib, libchronoEngine_postprocess.dylib and libchronoEngine_irrlicht.dylib it has hardcoded dependencies on these: 
```
> otool -L robottiexe                                                                                                  
robottiexe:
	/Users/au443759/source/into-cps-association/chronofmu/thirdparty/chrono_build/lib/libChronoEngine.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/au443759/source/into-cps-association/chronofmu/thirdparty/chrono_build/lib/libChronoEngine_irrlicht.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/au443759/source/into-cps-association/chronofmu/thirdparty/chrono_build/lib/libChronoEngine_postprocess.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

However, this is a problem when i.e. someone is sending their executable file and libraries, as @rainergericke did when helping me.
So what I did was change all of these hardcoded paths to use the dylibs within the same folder. The script to do this, for the case above is:
```
#!/bin/bash

FILE=$1
EXISTING_LIBRARY_PATH=$2
echo "File to replace dylibs: ${FILE}"
echo "Existing library path: ${EXISTING_LIBRARY_PATH}"

install_name_tool -change "${EXISTING_LIBRARY_PATH}/libChronoEngine.dylib" @rpath/libChronoEngine.dylib $FILE
install_name_tool -change "${EXISTING_LIBRARY_PATH}/libChronoEngine_irrlicht.dylib" @rpath/libChronoEngine_irrlicht.dylib $FILE
install_name_tool -change "${EXISTING_LIBRARY_PATH}/libChronoEngine_postprocess.dylib" @rpath/libChronoEngine_postprocess.dylib $FILE
install_name_tool -change "${EXISTING_LIBRARY_PATH}/libChronoEngine.dylib" @rpath/libChronoEngine.dylib libChronoEngine_irrlicht.dylib
install_name_tool -change "${EXISTING_LIBRARY_PATH}/libChronoEngine.dylib" @rpath/libChronoEngine.dylib libChronoEngine_postprocess.dylib
install_name_tool -id libChronoEngine.dylib libChronoEngine.dylib
install_name_tool -id libChronoEngine_irrlicht.dylib libChronoEngine_irrlicht.dylib
install_name_tool -id libChronoEngine_postprocess.dylib libChronoEngine_postprocess.dylib
install_name_tool -add_rpath "@executable_path/./" $FILE
install_name_tool -add_rpath "@rpath/./" $FILE
install_name_tool -add_rpath "@loader_path/./" $FILE
```

and it is used as: 
```
./dylibs_replace.sh robottiexe /Users/au443759/source/into-cps-association/chronofmu/thirdparty/chrono_build/lib
```
Now, running `otool -L robottiexe` again gives:
```
otool -L robottiexe
robottiexe:
	@rpath/libChronoEngine.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libChronoEngine_irrlicht.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libChronoEngine_postprocess.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)

closed 2020-08-06 by CThuleHansen